zephyrproject-rtos/zephyr #36606: Regression in udp socket performance from zephyr v2.3.0 to v2.6.0

vandita5, 2021-06-29:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
When moving from zephyr v2.3.0 to zephyr v2.6.0, I see worsening of performance for udp network stack. 
Description of the test I'm running:
I'm using an app modified from echo_server sample, with only ipv4 udp sockets running. This modified app waits for a message from client, once a message is received, it sends out dummy data of a given size in a loop, with a sleep after each iteration. If dummy data > mtu (1472 after overhead) then it is split into multiple packets. I'll include a snippet of the code at the end in additional notes. 

I use this app along with a python client, which also gives the throughput of data. I use Segger SystemView to get the time it takes to queue and send data and the cpu load experienced. I use these three parameters to denote performance. In all three criteria, zephyr v2.6.0 performed worse. I'm using the same configuration as well (included at the end).

Results seen:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. update echo_server code
2. checkout zephyr-v2.3.0 , run west update, run echo_server sample with a client
3. checkout zephyr-v2.6.0 , run west update, run echo_server sample with a client

**Expected behavior**
A clear and concise description of what you expected to happen.
Based on release notes for zephyr-v2.6.0, changes were made which reduced latency of transmission of data. Given this, I expected the time taken to send data out to reduce while I've seen it a little more than double as comapred to zephyr v2.3.0

Relevant release notes:
* Enabled running without TX or RX threads. By default, one RX thread andno TX thread is created. If userspace support is enabled, then one RX and oneTX thread are created. This improves the network transmit latency when apacket is sent from application.
* Changed to push highest priority net_pkt directly to driver when sending and ifthere is at least one TX thread.
* Changed to use k_fifo instead of k_work in RX and TX processing. This preventsk_work from accessing already freed net_pkt struct. This also improves the latencyof network packets when the data is passed between different network threads.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Showstopper. 
* This blocks us from moving to zephyr v2.6.0 which is needed for our project to get other dma and adc related bug fixes. 
* Also, crushes our hopes for imporved network performance which will lead to inability to use zephyr for our project as the throughput required increases further (we're already having to up the st chip used from f4 to f7 due to high cpu usage seen with zephyr's network stack at ~10mbps throughput).

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) Macos for zephyr v2.6.0 and windows 10 for zephyr v2.3.0
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used tags zephyr-v2.3.0 and zephyr-v2.6.0

**Additional context**
Add any other context about the problem here.
Code modification frome cho_Server sample:
```
static int process_udp(struct data *data)
{
  int ret = 0;
  bool print_debug_info_first_send = true;
  static uint8_t dummy_data[DUMMY_DATA_SIZE_BYTES] = { 0 };

  memset(dummy_data, 1, DUMMY_DATA_SIZE_BYTES);

  NET_INFO("Waiting for UDP packets on port %d (%s)...", MY_PORT, data->proto);

  struct sockaddr client_addr;
  socklen_t client_addr_len = sizeof(client_addr);

  int received = recvfrom(data->udp.sock, data->udp.recv_buffer,
                          sizeof(data->udp.recv_buffer), 0, &client_addr,
                          &client_addr_len);
  if (received < 0) {
    /* Socket error */
    NET_ERR("UDP (%s): Connection error %d", data->proto, errno);
    ret = -errno;
  } else {
    uint32_t tmp_len = 0;
    uint8_t *tmp_ptr = dummy_data;
    while (true) {
      tmp_len = DUMMY_DATA_SIZE_BYTES;
      tmp_ptr = dummy_data;
      do {
        ret = sendto(data->udp.sock, tmp_ptr, tmp_len, 0, &client_addr,
                     client_addr_len);
        if (ret < 0) {
          NET_ERR("UDP (%s): Failed to send %d", data->proto, errno);
          ret = -errno;
          return ret;
        }
        if (print_debug_info_first_send) {
          NET_INFO("sent %d remaining %d", ret, tmp_len);
          print_debug_info_first_send = false;
        }
        tmp_len = tmp_len - ret;
        tmp_ptr = tmp_ptr + ret;

        if (++data->udp.counter % 1000 == 0U) {
          NET_INFO("%s UDP: Sent %u packets", data->proto, data->udp.counter);
        }

      } while (tmp_len);

      k_msleep(SLEEP_TIME_MS);
    }
  }

  return ret;
}
```

Kconfig
```
# Generic networking options
CONFIG_NETWORKING=y
CONFIG_NET_UDP=y
CONFIG_NET_TCP=n
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_POSIX_MAX_FDS=6
CONFIG_NET_CONNECTION_MANAGER=y

# Kernel options
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y

# Logging
CONFIG_NET_LOG=y
CONFIG_LOG=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_PRINTK=y

# Network buffers
# As we were sending more data than we were receiving, tx count values
# have been increased while still keeping total count (rx + tx) the same
# as zephyr sample echo-server. More experimentation needs to be done to
# find ideal values for these variables.
CONFIG_NET_PKT_RX_COUNT=10
CONFIG_NET_PKT_TX_COUNT=22
CONFIG_NET_BUF_RX_COUNT=10
CONFIG_NET_BUF_TX_COUNT=118
CONFIG_NET_CONTEXT_NET_PKT_POOL=y
CONFIG_NET_BUF_DATA_SIZE=256

# IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=10

# Network shell
CONFIG_NET_SHELL=y
CONFIG_SHELL=y

# Network application options and configuration
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"

# Number of socket descriptors might need adjusting
# if there are more than 1 handlers defined.
CONFIG_POSIX_MAX_FDS=12

# How many client can connect to server simultaneously
CONFIG_NET_SAMPLE_NUM_HANDLERS=1

# Tracing
# CPU usage
CONFIG_TRACING=y

CONFIG_SEGGER_SYSTEMVIEW=y

# Network related
CONFIG_NET_STATISTICS=y
```

Additonal screenshots from sysview
sending 1400 bytes followed by a 2ms sleep.
with zephyr v2.6.0
[image]

with zephyr v2.3.0
[image]

vandita5, 2021-06-29:
@jukkar tagging you to see if you've seen similar performance with new zephyr v2.6.0. please let me know who I could reach out to instead if you're not the right person for this. thanks!

jukkar, 2021-06-29:
v2.3 is quite old already and there has been changes to network stack regarding v2.6. Some of the changes should improve the latency in TX side, like not creating TX thread and using k_fifo instead of k_work. But it is certainly possible that some other change is causing bigger slowdown between these two releases. Are you able to bisect this by running your workload in 2.4 and 2.5 releases? Then it might be easier to try to pinpoint the culprit.

There are couple of settings that can show some info about how fast the packet traverses the network stack (from application to driver).
```
CONFIG_NET_PKT_TXTIME_STATS
CONFIG_NET_PKT_TXTIME_STATS_DETAIL
```
Then the `net stats` command will show some information about this. Unfortunately these options are not available in 2.3, but just wanted to mention those in order to help with debugging efforts.

cc @rlubos

vandita5, 2021-06-29:
@jukkar That is useful info. Thanks! I do see a difference in the threads created (no tx_workq for v2.6) but was surprised by the tx time being higher. Do you have any suspiscions on what these changes could be that are causing a worse performance?

Yes, I can run a similar test on the other releases. Do you think this test is a good check for performance? For us, achieveing ~12-14Mbps at cpu util <15% with an F7 is the most important. We see this throughput achieved at ~30% with the F7 with our current app (v2.3).

I will try and use those kconfig options to get the packet traversal time though the sysview screenshots should also give a similar info (from app to being sent). I haven't used the net shell yet but hopefully it'll work out of the box.

rlubos, 2021-06-29:
I think it would be good to figure out first, whether the decreased performance is due to changes in the network stack itself, or rather the ethernet driver you use. From the screenshots I assume it's eth_stm32 driver, and there seems to be quite a lot of activity since v2.3.0 (see https://github.com/zephyrproject-rtos/zephyr/commits/v2.6.0/drivers/ethernet/eth_stm32_hal.c, commits after May). Specifically there are changes in `eth_tx()` which might be worth checking. 

It might be tricky however to tell the difference with `NET_TC_TX_COUNT` set to 0, where both the net stack and the driver are processed from the same thread, so perhaps it'd be good tO set `CONFIG_NET_TC_TX_COUNT=1` so that the L2 and the driver are processed by a  separate thread.

vandita5, 2021-06-29:
@rlubos that's a good suggestion. I tried that kconfig option and see that while the tx workq time (time spent by ethernet hal to send data) has also increased ~47% (from 0.855ms to 1.26ms), the time in the `send` function has increases by around 150% (0.855ms to 2.19ms). I'll look through the commits as well.
Next, I'll try older versions v3.4 and v3.5 as @jukkar suggested.
Does this regression signify something that the zephyrproject team gives priority to?

Relevant screenshots with results seen

[image]

sysview for zephyr v2.6 sending 2800 bytes with a 2ms sleep in between. tc_tx_count set to 1.
[image]

sysview for zephyr v2.6 sending 2800 bytes with a 2ms sleep in between. tc_tx_count default to 1.
[image]

dleach02, 2021-06-29:
@vandita5 I believe Jukka is recommending you use 'git bisect' to see if you can locate a particular change between 2.3 and 2.6 where the performance shifted in your environment. The 'git bisect' will do a binary search to find the commit that most impacted your performance. 

The flow would be:

```
$ git bisect start
$ git bisect bad   # current version
$ git bisect good v2.3.0
$ west update
# run your test
$ git bisect <good/bad>   # depending on your results
# repeat 'west update', test, 'git bisect <good/bad>' till the process identifies a sha

```
on each step, just make sure you do a 'west update' to ensure the modules are in sync with the SHA.

rlubos, 2021-06-30:
I did some testing on the HW I have (nRF52840 with 802.15.4 L2), with `CONFIG_NET_TC_TX_COUNT=1` and a custom app (a simple loop periodically sending UDP packet to the void). I do observe a 50% increased processing time of the `sendto()` function between 2.5.0 and 2.6.0 (I've measured system ticks, an average `sendto()` processing time increased from 10 to 15). I didn't observe any significant change between 2.3.0 and 2.5.0.

There isn't a single commit to blame that caused the increase in the processing time. The major impact in my case seemed to have the net_if mutex protection (24b49f4399702b52281766a79f7e31a99c281c9c), but there are others as well that caused smaller processing time increase (545fb92fc1428be2a71af6c0a8a0b85187521744, 97bf53fdcf06c1c6537df8518839729bf8ec2757, 924e433ffbc863d9daecd633992e975bfd6addc3). I wouldn't call any of the aforementioned commits a regression though, all of them aim to increase the networking stack reliability (which comes at a cost of extra CPU cycles as it seems). So I don't really see what could be done to restore the previous state of affairs (given that I'd rather consider lack of a proper access protection a more serious issue than increased processing time).

vandita5, 2021-06-30:
@rlubos thanks for testing on your end. I tested with just the releases as well, and saw similar results. All the above results were with our custom board which has similar performance to nucleo f429zi (both based on stm32f429zi). Here's what I've seen in summary with the different releases between v2.3 and v2.6

[image]

Thanks for sharing the changes that could have caused this increase and also the general philosophy behind them. Given the processing time, we've decided to stick to zephyr v2.3 at this time and might reevaluate later. IMO (and I'm definitely biased given my application :) ) increased processing time should be considered more seriously as well as it is important when zephyr is used in high performance devices aiming to send a bunch of data out. Especially when we're dealing with single core embedded systems, we can't afford ~80% cpu usage when sending 10Mbps (this was with zephyr v2.3).

I'll close out this issue as we're deciding to stay with zephyr v2.3 for the foreseeable future. Thank you for your help folks! :)

closed 2021-06-30 by vandita5