zephyrproject-rtos/zephyr #36608: Unable to compile USB console example with uart_mux

ycsin, 2021-06-29:
**Describe the bug**
Unable to compile USB console example with uart_mux enabled. Compiler complaining repeated variable declaration by  [cdc_acm.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/usb/class/cdc_acm.c#L1104) & [uart_mux.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/console/uart_mux.c#L881)

```
zephyrproject/zephyr/drivers/console/uart_mux.c:897: multiple definition of `rx_ringbuf_0'; 
zephyr/libzephyr.a(cdc_acm.c.obj):C:/Users/yongc/zephyrproject/zephyr/subsys/usb/class/cdc_acm.c:1127: first defined here

zephyrproject/zephyr/drivers/console/uart_mux.c:897: multiple definition of `tx_ringbuf_0'; 
zephyr/libzephyr.a(cdc_acm.c.obj):C:/Users/yongc/zephyrproject/zephyr/subsys/usb/class/cdc_acm.c:1127: first defined here
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build [usb/console](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/usb/console) with `CONFIG_GSM_MUX` set
2. See compilation error

**Expected behavior**
Should compile fine (?)

**Impact**
Unable to use usb/console with gsm_mux

**Logs and console output**
Full compilation log: [usb_compilation_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/6729988/usb_compilation_log.txt)

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: gnuarmemb
 - Zephyr 2.6 RC2
 - stm32g0b1re custom board

**Additional context**
1. Using offline WIP usb driver
2. Can be workaround in the meantime by renaming either file's `&t(r)x_ringbuf_##x` to `&yyy_t(r)x_ringbuf_##x`

ycsin, 2021-06-29:
Pinging codeowners @jukkar @jfischer-no

ycsin, 2021-07-02:
#36617 Fixed this issue.

jfischer-no, 2021-08-25:
close it in favor of #37926

PR #37926 (2021-08-30): usb: cdc_acm: prefix ringbuffers with cdc_acm
Prefix ringbuffers with cdc_acm because it is not possible
to use static specifier with RING_BUF_*_DECLARE_* macros
to avoid conflicts.

Fixes: #36608
commit 9d455817 (2021-08-30): usb: cdc_acm: prefix ringbuffers with cdc_acm

Prefix ringbuffers with cdc_acm because it is not possible
to use static specifier with RING_BUF_*_DECLARE_* macros
to avoid conflicts.

Fixes: #36608
closed 2021-08-30 by nashif