zephyrproject-rtos/zephyr #36609: could not mount fatfs on efm32pg_stk3402a

allen-k-liu, 2021-06-29:
**Describe the bug**
What have you tried to diagnose or workaround this issue? I just printout message after f_mount called in fatfs_mount.
Vstatic int fatfs_mount(struct fs_mount_t *mountp)
{
	FRESULT res;

	res = f_mount((FATFS *)mountp->fs_data, &mountp->mnt_point[1], 1);
	printf("fatfs_mnt: fs_data %p, mp[0]: %s; res = %d\n", mountp->fs_data, mountp->mnt_point, res);

**To Reproduce**
Steps to reproduce the behavior:
west build -b efm32pg_stk3402a samples/subsys/shell/fs/
[3/9] Linking C executable zephyr/zephyr_prebuilt.elf

[9/9] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       89172 B         1 MB      8.50%
            SRAM:       35264 B       256 KB     13.45%
        IDT_LIST:          0 GB         2 KB      0.00%

**Expected behavior**
I expect f_mount return 0

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Cannot use fatfs

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

uart:~$ *** Booting Zephyr OS build v2.6.0-rc2-119-g41d70790cd49  ***
[00:00:00.000,000] <inf> flash_gecko: Device FLASH_CTRL initialized
[00:00:00.000,000] <inf> fs: fs register 1: 0
[00:00:00.000,000] <inf> littlefs: littlefs registered**
[00:00:00.005,000] <inf> fs: fs register 0: 0
uart:~$ fs mount fat /fa
fatfs_mnt: fs_data 0x20004d34, mp[0]: /fa; res = 3 (FR_NOT_READY)  \printf("fatfs_mnt: fs_data %p, mp[0]: %s; res = %d\n", mountp->fs_data, mountp->mnt_point, res);\\
Error mounting fat fs.Error Code [-5]
[00:00:27.839,000] <inf> disk: disk_access, going return
[00:00:27.839,000] <err> fs: fs mount error (-5)

uart:~$ fs mount littlefs /lfs
[00:08:19.582,000] <inf> littlefs: LittleFS version 2.2, disk version 2.0; area_id = 0
[00:08:19.582,000] <inf> littlefs: FS area 0(devid = 0) at 0xfd800 for 10240 bytes
[00:08:19.583,000] <inf> littlefs: FS at FLASH_CTRL:0xfd800 is 5 0x800-byte blocks with 512 cycle
[00:08:19.583,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:08:19.585,000] <inf> littlefs: /lfs mounted

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) zyphyr
 - Toolchain (e.g Zephyr SDK, ...)  west CMake
 - Commit SHA or Version used
 git log
commit 41d70790cd49df7b0867a3a31485d8691f6b75bb (HEAD, origin/main)
Author: Jimmy Johnson <james.johnson672@t-mobile.com>
Date:   Wed Jun 23 22:01:18 2021 +0000

**Additional context**
Add any other context about the problem here.
I have similar problem for samples/subsys/fs/fat_fs/

allen-k-liu, 2021-06-29:
I'm sorry that I used modified version for zyphyrproject. Now I used earlier version:
al@al-ThinkPad-T480s:~/f_mount_rep/zephyrproject/zephyr$ git log
commit 20896ba1e09a3042edb953dc25750ba3405502c1 (HEAD)
Author: Anas Nashif <anas.nashif@intel.com>
Date:   Thu May 13 06:05:01 2021 -0400

    thread analyzer: fix formatting for unsigned int
    
    On 64bit systems, this was issuing warnings due to wrong formatting.
    
    Signed-off-by: Anas Nashif <anas.nashif@intel.com>

add fat element in menuconfig:
subsystem & os services -> File sys -> ELM FAT file sys support

and save it and build samples/subsys/shell/fs
the consle output:

[00:00:00.000,000] <inf> flash_gecko: Device FLASH_CTRL initialized
uart:~$ 
uart:~$ 
uart:~$ 
uart:~$ fs mount fat /D0
Error mounting fat fs.Error Code [-5]
[00:00:15.615,000] <err> fs: fs mount error (-5)

chrta, 2021-07-21:
I am not sure what the requirements are to use elm fat file system in zephyr. On the board there is no external memory available to the EFM32PG MCU and no SDCard slot. So only the internal flash memory could be used by default.

@allen-k-liu Do you want to use fat fs in a partition of the internal flash or do you have some external memory connected?

chrta, 2021-07-21:
@de-nordic Maybe you can help: What is required to get the fat fs sample running or can you can point to the relevant documentation/sample code?

de-nordic, 2021-07-21:
There is basic fatfs in internal/external memory sample here: https://github.com/zephyrproject-rtos/zephyr/pull/33252 that is still in PR state.

closed 2021-10-05 by github-actions[bot]