zephyrproject-rtos/zephyr #36613: LoRaWAN - Provide method to register a callback for received data

ballardr, 2021-06-29:
**Is your enhancement proposal related to a problem? Please describe.**
LoRaWAN is intended to allow downstream payloads being sent from the LoRaWAN server. However, it seems this has not been implemented yet in zephyr. See lines 101-107 of  $(ZEPHYR_BASE)/subsys/lorawan/lorawan.c
```
	if (mcpsIndication->RxData == true) {
		if (mcpsIndication->BufferSize != 0) {
			LOG_DBG("Rx Data: %s",
				log_strdup(mcpsIndication->Buffer));
		}
	}
```

**Describe the solution you'd like**
Similar to issue #36543, this should probably be implemented with a __weak function and allow the application developer to make their own non weak function to override, and process downstream payloads.

**Describe alternatives you've considered**
A CONFIG_ which sets the function or using callback registering functions

**Additional context**
For reference, mcpsIndication->RxData, mcpsIndication->Buffer and mcpsIndication->BufferSize are defined in LoRaMac.h (loramac-node module) as:
```
    /*!
     * Indicates, if data is available
     */
    bool RxData;
    /*!
     * Pointer to the received data stream
     */
    uint8_t* Buffer;
    /*!
     * Size of the received data stream
     */
    uint8_t BufferSize;
```

ballardr, 2021-06-29:
@Mani-Sadhasivam I think you are the main contributor for the LoRaWAN subsystem so this may be you?

tagunil, 2021-07-01:
As far as I can tell, the downstream callbacks are already implemented in Zephyr 2.6.

ballardr, 2021-07-01:
Ah yes, I didn't see that in the release notes (and have been using 2.5). Closing now

closed 2021-07-01 by ballardr