zephyrproject-rtos/zephyr #36625: compilation fails while building samples/net/openthread/coprocessor for Arduino nano 33 ble

nageshshamnur, 2021-06-29:
**Describe the bug**
compilation fails when trying to build openthread/coprocessor as explained in the page for arduino nano 33 ble

**To Reproduce**
Steps to reproduce the behavior:
1. ```$ west build -p auto -b arduino_nano_33_ble samples/net/openthread/coprocessor -- -DCONF_FILE="prj.conf overlay-rcp.conf"```
2. see error as below:
```
$ west build -p auto -b arduino_nano_33_ble samples/net/openthread/coprocessor -- -DCONF_FILE="prj.conf overlay-rcp.conf"
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/ohos-bld/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/ohos-bld/zephyrproject/zephyr/samples/net/openthread/coprocessor
-- Zephyr version: 2.6.99 (/home/ohos-bld/zephyrproject/zephyr), build: zephyr-v2.6.0-573-gf6a7dce763e8
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: arduino_nano_33_ble
-- Cache files will be written to: /home/ohos-bld/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.3 (/home/ohos-bld/zephyr-sdk-0.12.3)
-- Found dtc: /home/ohos-bld/zephyr-sdk-0.12.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/ohos-bld/zephyrproject/zephyr/boards/arm/arduino_nano_33_ble/arduino_nano_33_ble.dts
-- Generated zephyr.dts: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/ohos-bld/zephyrproject/zephyr/samples/net/openthread/coprocessor/Kconfig
Loaded configuration '/home/ohos-bld/zephyrproject/zephyr/boards/arm/arduino_nano_33_ble/arduino_nano_33_ble_defconfig'
Merged configuration 'prj.conf'
Merged configuration 'overlay-rcp.conf'
Configuration saved to '/home/ohos-bld/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/ohos-bld/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'

warning: UART_CONSOLE_ON_DEV_NAME (defined at drivers/console/Kconfig:12) was assigned the value
'UART_1' but got the value ''. Check these unsatisfied dependencies: (UART_CONSOLE ||
CONSOLE_SUBSYS) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_CONSOLE_ON_DEV_NAME.html and/or
look up UART_CONSOLE_ON_DEV_NAME in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

warning: LOG_BACKEND_RTT (defined at subsys/logging/Kconfig.backends:88) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: USE_SEGGER_RTT (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_LOG_BACKEND_RTT.html and/or look up
LOG_BACKEND_RTT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ohos-bld/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Deprecation Warning at /home/ohos-bld/zephyrproject/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- OpenThread Source Directory: /home/ohos-bld/zephyrproject/modules/lib/openthread
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Searching 16 bit integer - Using unsigned short
-- Check if the system is big endian - little endian
[... 62 lines cut ...]
/home/ohos-bld/zephyrproject/zephyr/drivers/ieee802154/ieee802154_nrf5.c:468:3: warning: implicit declaration of function 'nrf_802154_retransmit_csma_ca_raw'; did you mean 'nrf_802154_transmit_csma_ca_raw'? [-Wimplicit-function-declaration]
  468 |   nrf_802154_retransmit_csma_ca_raw(payload);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   nrf_802154_transmit_csma_ca_raw
[163/380] Building C object zephyr/subsys/net/l2/openthread/CMakeFiles/subsys__net__ip__l2__openthread.dir/openthread.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/ohos-bld/zephyrproject/zephyr/build
```

**Expected behavior**
compilation is ok.

**Impact**
Unable to build openthread RCP example on Arudino Nano 33 BLE

**Logs and console output**
as attached above.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.2 LTS
 - Toolchain: zephyr-sdk-0.12.3
 -  Commit SHA: f6a7dce763e8dcdcdda6d9977fd327386f554f98

**Additional context**
git revision: 457a28bf78 was successful earlier.

nageshshamnur, 2021-06-30:
with further debugging found that, the following piece of code might not be supported by Arduino nano 33 ble which is also powered by nRF52840 SoC.

file: ./drivers/ieee802154/ieee802154_nrf5.c inside the function: nrf5_get_capabilities_at_boot( ) parameters --> ```NRF_802154_CAPABILITY_DELAYED_RX``` and ```NRF_802154_CAPABILITY_SECURITY```

file: ./drivers/ieee802154/ieee802154_nrf5.c function: function --> ```nrf_802154_retransmit_csma_ca_raw( )```

rlubos, 2021-06-30:
Are your repositories in sync, i. e. did you call `west update`? The issue looks as if the radio driver you build with was outdated, I cannot reproduce the failure on the current master.

nageshshamnur, 2021-06-30:
i had run ```west update`` earlier, but running again now seems to have solved my problem. Sorry for the trouble caused.

closed 2021-06-30 by nageshshamnur

rlubos, 2021-06-30:
> i had run `west update` earlier, but running again now seems to have solved my problem. Sorry for the trouble caused.

No worries, it happens quite often if you're not used to using west.