zephyrproject-rtos/zephyr #36631: Turn on GPIO from DTS

novalisek, 2021-06-29:
**Is your enhancement proposal related to a problem? Please describe.**
Hi, I would like to turn on GPIO from DTS, I need to turn/hold ON "enable" pin of DC/DC converter as soon as firmware starts

**Describe the solution you'd like**
To be able to use something like
```
 outputs {
       compatible = "gpio-leds";
		dcdc: dcdc {
        	gpios = <&gpio0 18 (GPIO_ACTIVE_HIGH|GPIO_OUTPUT_HIGH)>;
       };   	 		        
    };     
```

When I try to add GPIO_OUTPUT_HIGH, I get DTS errors.

**Additional context**
I need this for MCUBOOT - I can turn on GPIO manualy by gpio_pin_set() in my firmware, but I also need to do it in MCUBOOT, so that DC/DC stays ON when user releases power-on button, because it takes few seconds if you swap firmware and I don't want DC/DC to be turned OFF while swapping, if user releases the button earlier :)

**Describe alternatives you've considered**
I could modify MCUBOOT source to use gpio_pin_set(), but I would better not to touch MCUBOOT source code and use DTS for this.

May be this  feature is already possible and I am just doing something wrong ... 

Thanks

galak, 2021-06-30:
@novalisek I think you just need in the board port that builds MCUBOOT to call `gpio_pin_set()`.

novalisek, 2021-06-30:
@galak what do you mean by "board port"? I am just using my own  dts overlay. thanks

galak, 2021-06-30:
You are building for some board correct?  How are you getting dcdc set right now?

novalisek, 2021-06-30:
I am using nrf52dk_nrf52832 as base with my overlay.  In my firmware I normaly use `gpio_pin_set` to switch EN pin of external DCDC. 
I would like the EN pin also to be switched automaticaly immediately when mcuboot starts (idealy without modifying bootloader\mcuboot\boot\zephyr\main.c)

galak, 2021-06-30:
> I am using nrf52dk_nrf52832 as base with my overlay. In my firmware I normaly use `gpio_pin_set` to switch EN pin of external DCDC.
> I would like the EN pin also to be switched automaticaly immediately when mcuboot starts (idealy without modifying bootloader\mcuboot\boot\zephyr\main.c)

I'm not an expert w/regards to mcuboot, but its my understanding that the mcuboot build would pull in part of zephyr and I assume that includes some board code.  So in theory you should be able to add something in boards/arm/nrf52dk_nrf52832 that would call gpio_pin_set() and that should than happen in both mcuboot and zephyr.

galak, 2021-06-30:
@de-nordic, @d3zd3z I'm guessing you might be able to answer @novalisek question better than I can.

closed 2021-06-30 by galak