zephyrproject-rtos/zephyr #36646: sample.shell.shell_module.minimal_rtt fails to build on mimxrt1170_evk_cm4/mimxrt1170_evk_cm7

galak, 2021-06-30:
`./scripts/twister -p mimxrt1170_evk_cm4 -p mimxrt1170_evk_cm7 -s samples/subsys/shell/shell_module/sample.shell.shell_module.minimal_rtt`

Fails as follows:

```
[ 91%] Linking C static library lib..__modules__hal__nxp.a
[ 91%] Built target ..__modules__hal__nxp
Scanning dependencies of target modules__segger
[ 92%] Building C object modules/segger/CMakeFiles/modules__segger.dir/home/galak/git/modules/debug/segger/SEGGER/SEGGER_RTT.c.obj
[ 92%] Building C object modules/segger/CMakeFiles/modules__segger.dir/SEGGER_RTT_zephyr.c.obj
[ 93%] Linking C static library libmodules__segger.a
[ 93%] Built target modules__segger
Scanning dependencies of target zephyr_prebuilt
[ 94%] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
[ 95%] Linking C executable zephyr_prebuilt.elf
/opt/zephyr-sdk-0.13.0-alpha-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: warning: orphan section `.dtcm_data' from `../modules/segger/libmodules__segger.a(SEGGER_RTT.c.obj)' being placed in section `.dtcm_data'
collect2: error: ld returned 1 exit status
gmake[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:122: zephyr/zephyr_prebuilt.elf] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:3062: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
gmake: *** [Makefile:103: all] Error 2
```

galak, 2021-06-30:
Its possible we need some special handling for RT117x in:

`soc/arm/nxp_imx/rt/Kconfig.defconfig.series` for:

```
choice SEGGER_RTT_SECTION
        default SEGGER_RTT_SECTION_DTCM if HAS_MCUX_SEMC
        depends on USE_SEGGER_RTT
endchoice

choice SEGGER_SYSVIEW_SECTION
        default SEGGER_SYSVIEW_SECTION_DTCM if HAS_MCUX_SEMC
        depends on SEGGER_SYSTEMVIEW
endchoice
```

dleach02, 2021-07-02:
So this fails using twister but if I use west with the '-T sample.shell.shell_module.minimal_rtt' for both targets it will build/link without errors.

`west build -b mimxrt1170_evk_cm4 -p auto -T sample.shell.shell_module.minimal_rtt samples/subsys/shell/shell_module/`

PR #36704 (2021-07-07): tracing: debug: Link Segger RTT/SystemView sections in DTCM if chosen
Configures Segger RTT and SystemView data linker sections to DTCM by
default if there is a zephyr,dtcm chosen node in devicetree. This fixes
a build warning in samples/subsys/shell/shell_module for the
mimxrt1170_evk_cm7 and mimxrt1170_evk_cm4 platforms, which don't
currently have a zephyr,dtcm chosen node.

Note that there are runtime issues with Segger RTT and SystemView on
this board that need further debug, but submitting this patch now to
address nightly CI failures.

Fixes #36646

@erwango This change affects some ST b [... cut]
closed 2021-07-07 by nashif