zephyrproject-rtos/zephyr #36649: Adding Clock and Clock Generator Options for atmel_sam0

photonthunder, 2021-06-30:
I am interested in improving the atmel sam0 drivers so they provide better ability to adjust the clock.  The challenge with this set of chips (I have worked on the samd21 and the samd51) is that you have a variety of clock sources (crystals, oscillators, DFLL, DPLL) that can run at a variety of frequencies and then you tie those to clock generators that can be connected to the different peripherals.  In my current design I run off a 20 MHz external oscillator but run my peripherals at 5 MHz.  The current code uses CONFIGS added to the fork of the zephyr code.  The hard part is if you tried to add CONFIGS for all options it will explode.  One option would be to do like what is done in pinmux.c where you set the appropriate pin/peripheral mux.  This isn't ideal but it is better than making changes to the zephyr code and ties it to the board.  I am sure this has been discussed before so just curious what people currently do and if there is any solid options for handling the clock options?

trunifom, 2021-08-03:
I'm working with the nRF53 microcontroller from Nordic. 

#------------------------------------------
# Clock: Enable Internal 32kHz Clock
#------------------------------------------
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=n
CONFIG_CLOCK_CONTROL_NRF_K32SRC_XTAL=n
CONFIG_CLOCK_CONTROL_NRF_K32SRC_SYNTH=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_20PPM=y
CONFIG_CLOCK_CONTROL_NRF_CALIBRATION_DEBUG=y

trunifom, 2021-08-03:
I'm working with the nRF53 microcontroller from Nordic.
When I was working with the DevKit, everything was running fine. When I switched to my own hardware (customboard), suddenly nothing worked anymore (when I wasn't debugging). The problem was that I didn't have an external 32kHz clock equipped.

Nordic also solves the problem with configs.

#------------------------------------------
# Clock: Enable Internal 32kHz Clock
#------------------------------------------
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=n
CONFIG_CLOCK_CONTROL_NRF_K32SRC_XTAL=n
CONFIG_CLOCK_CONTROL_NRF_K32SRC_SYNTH=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_20PPM=y
CONFIG_CLOCK_CONTROL_NRF_CALIBRATION_DEBUG=n

trunifom, 2021-08-03:
I'm working with the nRF53 microcontroller from Nordic.
When I was working with the DevKit, everything was running fine. When I switched to my own hardware (customboard), suddenly nothing worked anymore (when I wasn't debugging). The problem was that I didn't have an external 32kHz clock equipped. The solution was the configurations.

So Nordic seems to solve this problem via configs

CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=n
CONFIG_CLOCK_CONTROL_NRF_K32SRC_XTAL=n
CONFIG_CLOCK_CONTROL_NRF_K32SRC_SYNTH=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_20PPM=y
CONFIG_CLOCK_CONTROL_NRF_CALIBRATION_DEBUG=n

photonthunder, 2021-08-03:
@truniform, thanks for the input.  That is one method and I do something similar in my hacks fo the sam0 code (so I can activate my 20 MHz external oscillator and setup GCLK3 as a 5 MHz divisor.  This is where the samd/same family of atmel/microchip processors get hard to "setup".  You would have configs for system clocks, then configs to setup the GCLK, and then configs to tell the peripheral which GCLK to connect.   I think the best way would be to do it in the device tree so you could account for all the options, but it would be a lot of work to try and make it all inclusive and it wouldn't really scale out to other processors so was curious what others had done or if there was even interest in the effort.  It may be that just doing some custom hacking of the drivers is the best option?