zephyrproject-rtos/zephyr #36655: twister:  sometimes the twister fails because the error `configparser.NoSectionError: No section: 'manifest'`

wangnuannuan, 2021-07-01:
**Describe the bug**

Sometimes the twister  fails because the error `configparser.NoSectionError: No section: 'manifest'`
```
CMake Error at /slowfs/ru20arcjenkins/de02-arcjenkins2/slaves/ru20-gp-srv-linux03/workspace/arcoss_verification/sandbox_jingru/testing_jobs/zephyr_verification/sanity_check/zephyr/cmake/zephyr_module.cmake:63 (message):
  Traceback (most recent call last):

    File "/depot/Python/Python-3.7.0/lib/python3.7/site-packages/west/manifest.py", line 172, in _mpath
      path = cp.get('manifest', 'path')
    File "/depot/Python/Python-3.7.0/lib/python3.7/configparser.py", line 780, in get
      d = self._unify_values(section, vars)
    File "/depot/Python/Python-3.7.0/lib/python3.7/configparser.py", line 1146, in _unify_values
      raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'manifest'
```
That's because the specific platform will call `west` command, which will try to modify the config file `.west/config`. Take `nsim_hs_smp` as example, the `twister` will call `west` command at https://github.com/zephyrproject-rtos/zephyr/blob/882600834e80d2d3d5db4e61d72f63a8bd997fcc/scripts/pylib/twister/twisterlib.py#L520.

The contents of config are:
```
[manifest]
path = zephyr-downstream
file = west.yml
```
https://github.com/zephyrproject-rtos/west/blob/main/src/west/app/main.py#L406. The `west flash` will modify it's contents as:

```
[manifest]
path = zephyr-downstream
file = west.yml
[zephyr]
base = zephyr
```
Because the twister run this command with multiple processes, so I guess there are race when ·twister· call this command, then the file is not modified successfully due to multi-process race.

I found that `west` tool will update the contents of `.west/config` at 
1. west extension commands(build, flash...) https://github.com/zephyrproject-rtos/west/blob/21a478e3c87f6c910aa8b293ee50f7c8480273e6/src/west/app/main.py#L703, this may affect twister
2.  west init command https://github.com/zephyrproject-rtos/west/blob/main/src/west/app/project.py#L230, https://github.com/zephyrproject-rtos/west/blob/main/src/west/app/project.py#L299
3. west config command

cfriedt, 2021-08-06:
@nashif - looks like this was not fixed in the linked PR - what's happening?

nashif, 2021-09-16:
fixed in #38325

closed 2021-09-16 by nashif