zephyrproject-rtos/zephyr #36659: samples/net/sockets small bugs

maksimdrachov, 2021-07-01:
**Describe the bug**
Some examples contain some small bugs.

_**samples/net/sockets/pocket
samples/net/sockets/tcp**_
[image]
```
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
```
needs to be added to prj.conf for it to work

_**samples/net/sockets/echo_client**_
Doesn't work, can't figure out how to fix.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr/samples/net/sockets/NAME-EXAMPLE
2. west build -b nucleo_f756zg
3. See error

**Expected behavior**
A successful build

**Impact**
Annoyance, but can be easily fixed in the case of the first two.
Don't have a lot of experience with Zephyr, so need someone else to take a look at the third one.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain Zephyr
 - Commit SHA or Version used
```
VERSION_MAJOR = 2
VERSION_MINOR = 6
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =
```

nashif, 2021-07-06:
Can you please submit a PR to fix those small issues?

rlubos, 2021-07-12:
I'm not able to reproduce the third failure (the `echo_client` one), perhaps you need to run a clean build?

rlubos, 2021-07-26:
#36860 with a fix for the first two problems was merged. As the third one doesn't seem to be reproducible, I'm closing the issue. Please reopen if it still persists somehow (but as mentioned earlier it likely just needs a clean build).

closed 2021-07-26 by rlubos