zephyrproject-rtos/zephyr #36665: l2cap cids mixed up in request

krjmelik, 2021-07-01:
**Describe the bug**
When peripheral device wants to disconnect it sends request with scid and dcid. But it sends rx.cid for dcid and tx.cid for scid. And according to Bluetooth specification it is wrong.

**To Reproduce**
1. Open l2cap connection between device configured as peripheral to a phone as central.
2. Send disconnect request from device.
3. See wrong cids in the request message.

**Expected behavior**
int bt_l2cap_chan_disconnect(struct bt_l2cap_chan *chan)
{ ...
req->dcid = sys_cpu_to_le16(ch->tx.cid);
req->scid = sys_cpu_to_le16(ch->rx.cid);
...}

**Impact**
Wrong order of cid leads to:
central device potentially closes its wrong cid;
central device sends response with the same wrong order which leads to attempt to close tx.cid (which doesn't exist) instead of rx.cid on the peripheral device.

**Environment:**
 - OS: Windows
 - Toolchain Zephyr
 - v2.5.0-rc3 (still exists in master)

**Additional context**
The issue appeared after #16771 and #16799, commit e61c5b3630ef013907c62acfdd53180a30cd5ecd. But I guess the fix was overdone. le_disconn_req and le_disconn_rsp are correct, only bt_l2cap_chan_disconnect is wrong. Even debug message is still correct.

carlescufi, 2021-07-15:
@krjmelik could you please send a Pull Request with the suggested fix in the description?

krjmelik, 2021-07-16:
Unfortunately, I don't have the access to my workplace now.
But as I said the only change is in: subsys/bluetooth/host/l2cap.c,
function: int bt_l2cap_chan_disconnect(struct bt_l2cap_chan *chan)

change this:
req->dcid = sys_cpu_to_le16(ch->rx.cid);
req->scid = sys_cpu_to_le16(ch->tx.cid);

to this:
req->dcid = sys_cpu_to_le16(ch->tx.cid);
req->scid = sys_cpu_to_le16(ch->rx.cid);

Because for the request it should be rx.cid for the source cid, and tx.cid to the destination cid.

jhedberg, 2021-07-19:
> Unfortunately, I don't have the access to my workplace now.

Ok, no problem, I submitted #37049 with this fix

PR #37049 (2021-07-21): Bluetooth: L2CAP: Fix CIDs in disconnect request
The destination and source CIDs were mixed up in the L2CAP
Disconnect request.

Fixes #36665
commit 9aedfa67 (2021-07-21): Bluetooth: L2CAP: Fix CIDs in disconnect request

The destination and source CIDs were mixed up in the L2CAP
Disconnect request.

Fixes #36665
closed 2021-07-21 by jhedberg