zephyrproject-rtos/zephyr #36667: logger: Filesystem backend doesn't work except for first time boot

ExtremeGTX, 2021-07-01:
**Describe the bug**
After activating `LOG_BACKEND_FS`, `CONFIG_FILE_SYSTEM_LITTLEFS` and their dependencies, it worked as expected on the very first time boot (clean storage partition).

The problem is after rebooting the board, nothing new got saved to the log file created from the first time boot nor a new file created. as if logging to filesystem is completely stopped.

**What have you tried to diagnose or workaround this issue?**

I tried to do some debugging, and I found that adding `fs_seek(&file, 0, FS_SEEK_END);` just before https://github.com/zephyrproject-rtos/zephyr/blob/882600834e80d2d3d5db4e61d72f63a8bd997fcc/subsys/logging/log_backend_fs.c#L177 make things work. but I think this is not the right way to do it.

After adding the workaround above, makes the logging backend create a new file on each reboot, instead of appending to the old one till it reaches the `CONFIG_LOG_BACKEND_FS_FILE_SIZE`. that also breaks the limit `CONFIG_LOG_BACKEND_FS_FILES_LIMIT`, I end up getting 3 files after 3 or more reboots
```
log.0000
log.000x
log.000(x+1)
```

**To Reproduce**
Steps to reproduce the behavior:
1. replace the content of `zephyr\samples\hello_world\src\main.c` with the following sample code:
```
	#include <zephyr.h>
	#include <sys/printk.h>
	
	#include <logging/log.h>
	LOG_MODULE_REGISTER(main, LOG_LEVEL_DBG);
		  
	  void main(void)
	  {
		  int x = 0;
		  while (x < 20)
		  {
			  k_sleep(K_MSEC(200));
			  LOG_INF("Hello World! %s %d", CONFIG_BOARD, x++);
		  }
	  }
 ```
2. west build -b stm32f769i_disco ./samples/hello_world
3. `west build -b stm32f769i_disco -t menuconfig` and use the attached .config below
4. flash

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
showstopper

**Environment (please complete the following information):**
 - Board STM32F769i_Disco
 - Windows
 - Zephyr SDK
 - main@https://github.com/zephyrproject-rtos/zephyr/commit/882600834e80d2d3d5db4e61d72f63a8bd997fcc

**Additional context**

<details>
<summary>My .config</summary>

```
CONFIG_GPIO=y
# CONFIG_SPI is not set
# CONFIG_I2C is not set
# CONFIG_KSCAN is not set
# CONFIG_WIFI is not set
# CONFIG_MODEM is not set
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_BOARD="stm32f769i_disco"
CONFIG_SOC="stm32f769xx"
CONFIG_SOC_SERIES="stm32f7"
CONFIG_NUM_IRQS=110
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=216000000
[... 1028 lines cut ...]
# CONFIG_BUILD_OUTPUT_S19 is not set
# CONFIG_BUILD_OUTPUT_UF2 is not set
# CONFIG_BUILD_OUTPUT_STRIPPED is not set
# CONFIG_APPLICATION_DEFINED_SYSCALL is not set
# CONFIG_MAKEFILE_EXPORTS is not set
# end of Build Options
# end of Build and Link Features

#
# Boot Options
#
# CONFIG_IS_BOOTLOADER is not set
# CONFIG_BOOTLOADER_MCUBOOT is not set
# CONFIG_BOOTLOADER_BOSSA is not set
# end of Boot Options

#
# Compatibility
#
CONFIG_COMPAT_INCLUDES=y
# end of Compatibility

```

</details>

ExtremeGTX, 2021-07-02:
@MSyc-NS Could you please help me understand, Which case is being handled by the condition at line 295 ? https://github.com/zephyrproject-rtos/zephyr/blob/3221a1e8e703f94593c021c3434e135d05797f57/subsys/logging/log_backend_fs.c#L295 
And why multiplying `CONFIG_LOG_BACKEND_FS_FILES_LIMIT` by `2` ?

elliot-wdtl, 2021-07-13:
I thought that it was expected to create a new file on each boot but I noticed that it treated file with index 0 differently to the rest.
https://github.com/zephyrproject-rtos/zephyr/pull/36946 addresses that but if it is intended to append to latest file on each boot additional changes will be required.

ExtremeGTX, 2021-07-14:
@elliot-wdtl I could share my changes, if it is acceptable to change the behavior to "append to the last file".
Also I implemented log rotate functionality which will keep filenames between `log.0000` and `log.CONFIG_LOG_BACKEND_FS_FILES_LIMIT`

PR #36946 (2021-07-17): logging: count file with index 0 as log file too
Fixes #36667

If you had a single log file with index 0 created and you reboot
the log backend wasn't counting it and was overwiting it.
If you filled that file up before rebooting then it worked as
expected, creating a new file at the next index on each boot.
commit 15318634 (2021-07-17): logging: count file with index 0 as log file too

Fixes #36667

If you had a single log file with index 0 created and you reboot
the log backend wasn't counting it and was overwiting it.
If you filled that file up before rebooting then it worked as
expected, creating a new file at the next index on each boot.
closed 2021-07-17 by cfriedt