zephyrproject-rtos/zephyr #36668: LittleFS example overwrite falsh memory

d-soliveira, 2021-07-01:
**Describe the bug**
I'm using the Samples/Littlefs with an STM32 EVB without any problems.
Now I want to save some data files (txt, html, json, etc) in the uC's flash, so I can read them during the code execution.
At this moment I can save the data directly to the memory (starting at 0x080FC000), using external tools (STM32CubeProgrammer), but when I upload the program the data that I have stored previously is overwritten as depicted in the figure.

**To Reproduce**
Steps to reproduce the behavior:
1. Earase the uC's flash
2. Save some known data into flash starting at 0x080FC000, or other addres defined in DTS file for store_partition
3. Verify the data was written
4. Upload the Littlefs example
5. Verify the data previously written was overwitten by the program.

**Expected behavior**
Is expected that the store_partition is not overwritten by the program during the build/upload process

**Impact**
At this moment isn't possible to use the internal flash to store data for later reading/writting.

**Logs and console output**
Screenshot
[image]

**Environment**
 - Linux
 - PlatformIO

de-nordic, 2021-07-01:
@d-soliveira If you share the flash at given address with LittleFS then it will format the area and write own super-block.

de-nordic, 2021-07-01:
@d-soliveira You need to disable mounting the LittleFS at storage partition if you want to be able to perform raw writes/read to it.

d-soliveira, 2021-07-02:
Hi, thank you for the answers.
If I understand, I have to upload my program first and only after that, in runtime, I can instruct the filesystem to store the files?
My idea was to save the files in flash at the same time I upload the program.

de-nordic, 2021-07-02:
You could do that but you would need to create image of LittleFS partition with the files, and write that image at specified location. If you just bare flash files, with no file system structure, the LittleFS driver recognized random data and thinks that the partition needs formatting.
There are tools to mount littlefs under various systems or create such file systems (https://github.com/littlefs-project/littlefs-fuse, https://pypi.org/project/littlefs-python/).

d-soliveira, 2021-07-02:
Thank you @de-nordic !

closed 2021-07-02 by d-soliveira