zephyrproject-rtos/zephyr #36678: Zephyr Throws Exception for Shell "log status" command when Telnet is shell backend

bob2oneil, 2021-07-01:
Zephyr throws an exception accessing a NULL entry when the system is configured to
target the UART as the log backend and the Telnet session for the shell, when the 
shell command "log status" is selected from the shell menu within the Telnet session
itself.

**To Reproduce**
Steps to reproduce the behavior:
1. Build a simple application with the kernel configuration as listed below
2. west build -p always -b native_posix_64
3. Run application
4. Establish a serial session using minicom to the pseudoterminal
5. Establish a telnet session to the primary Ethernet interfaces IP.
6. From with the Telnet session, invoke "log status" from the shell.

**Expected Behavior**
No exception, but log status shell menu directed to the Telnet session.
Other shell menus behave properly.

**Impact**
Annoyance, will need to be fixed.

**Logs and console output**
Attached is a screenshot using Visual Studio Code as the IDE when the 
exception occurs.  The exception occurs in the method "shell_state_precheck"
when attempting to access "shell->log_backend" which is 0.

**Environment (please complete the following information):**
 - OS: (e.g. Linux Ubuntu 20 POSIX build)
 - Toolchain (e.g Zephyr SDK, ...)
 - *** Booting Zephyr OS build zephyr-v2.6.0-550-gc08f3751bd07  ***

**Additional context**

```
# Shell and logging device backends
CONFIG_SHELL_LOG_BACKEND=n
CONFIG_SHELL_BACKEND_TELNET=y
CONFIG_LOG_BACKEND_UART=y
```

Callstack is as follows:

```
shell_state_precheck(const struct shell * shell) (/home/robocom/zephyrproject/zephyr/subsys/logging/log_cmds.c:58)
cmd_log_self_status(const struct shell * shell, size_t argc, char ** argv) (/home/robocom/zephyrproject/zephyr/subsys/logging/log_cmds.c:126)
exec_cmd(const struct shell * shell, size_t argc, const char ** argv, const struct shell_static_entry * help_entry) (/home/robocom/zephyrproject/zephyr/subsys/shell/shell.c:549)
execute(const struct shell * shell) (/home/robocom/zephyrproject/zephyr/subsys/shell/shell.c:779)
state_collect(const struct shell * shell) (/home/robocom/zephyrproject/zephyr/subsys/shell/shell.c:961)
shell_process(const struct shell * shell) (/home/robocom/zephyrproject/zephyr/subsys/shell/shell.c:1419)
shell_signal_handle(const struct shell * shell, enum shell_signal sig_idx, shell_signal_handler_t handler) (/home/robocom/zephyrproject/zephyr/subsys/shell/shell.c:1241)
shell_thread(void * shell_handle, void * arg_log_backend, void * arg_log_level) (/home/robocom/zephyrproject/zephyr/subsys/shell/shell.c:1302)
z_thread_entry(k_thread_entry_t entry, void * p1, void * p2, void * p3) (/home/robocom/zephyrproject/zephyr/lib/os/thread_entry.c:29)
posix_thread_starter(void * arg) (/home/robocom/zephyrproject/zephyr/arch/posix/core/posix_core.c:305)
libpthread.so.0!start_thread(void * arg) (/build/glibc-eX1tMB/glibc-2.31/nptl/pthread_create.c:477)
libc.so.6!clone() (/build/glibc-eX1tMB/glibc-2.31/sysdeps/unix/sysv/linux/x86_64/clone.S:95)

```

bob2oneil, 2021-07-01:
[image]

cfriedt, 2021-07-20:
@jakub-uC - was this fixed by a previous PR?

cfriedt, 2021-07-20:
@bob2oneil - can you please rebase and retest?

bob2oneil, 2021-07-20:
Yes, I will do so.  Thanksl

bob2oneil, 2021-07-20:
For pull on July 20th, 2020, problems appears to be fixed -- good job guys.

** Booting Zephyr OS build zephyr-v2.6.0-1206-g04ab2400e65c  ***

closed 2021-07-20 by bob2oneil