zephyrproject-rtos/zephyr #36679: Bluetooth - notifications not sending (bonded, CONFIG_BT_MAX_CONN=4, after disconnection then reconnection)

nwsetec, 2021-07-02:
**Describe the bug**
If our Zephyr v2.6.0 based Bluetooth peripheral device is reset and a mobile device connects (note it has previously been bonded) the mobile app can read data and subscribe to notifications.  If the mobile app is closed and reopened (Bluetooth is disconnected and reconnected) then the mobile app can now read characteristics but can no longer receive notifications (even when toggling the notification subscribe/unsubscribe button in the nRFConnect mobile app). If the Zephyr device is reset and the nRFConnect mobile app reconnects again then notifications will work.

Associated issues:
- https://github.com/zephyrproject-rtos/zephyr/issues/36381 (workaround apply in controller has resolve this issue for us)
- https://github.com/zephyrproject-rtos/zephyr/issues/36552

**To Reproduce**
See description from issue https://github.com/zephyrproject-rtos/zephyr/issues/36381

**Expected behavior**
Notifications always work.

**Impact**
Showstopper

**Logs and console output**
I'll add ASAP

**Environment (please complete the following information):**
See info from issue https://github.com/zephyrproject-rtos/zephyr/issues/36381

**Additional context**
Briefly discussed with @joerchan on Slack and https://github.com/zephyrproject-rtos/zephyr/issues/36552#issuecomment-872031943

cvinayak, 2021-07-02:
@nwsetec Has this worked before, if so, you could state the versions/commit it worked and label this as a regression.

rene-becker-setec, 2021-07-07:
@cvinayak - we have not used this configuration before, so can't say whether this is a regression or not (I am working with @nwsetec )

nwsetec, 2021-07-07:
> @nwsetec Has this worked before, if so, you could state the versions/commit it worked and label this as a regression.

We haven't had this working with this configuration unfortunately.  I removed BT SMP and authentication requirements off the characteristics and notifications seem to work ok.

carlescufi, 2021-08-05:
@nwsetec could you please provide full logs and also your `.config` file?

nwsetec, 2021-08-09:
@carlescufi 

My setup from the original issue: https://github.com/zephyrproject-rtos/zephyr/issues/36381:

**To Reproduce**
Setup a project the passive Bluetooth scanning with a 60ms scanning interval and 60ms window with and the filter duplicates disable and set up the project to allow up to 4 connected BLE devices and up to 4 bonds.

[prj.conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/6675076/prj.conf.txt)

``` c
#define BT_LE_SCAN_PASSIVE_ALLOW_DUPLICATES BT_LE_SCAN_PARAM(BT_HCI_LE_SCAN_PASSIVE, \
					    BT_HCI_LE_SCAN_FILTER_DUP_DISABLE, \
					    BT_GAP_SCAN_FAST_INTERVAL, \
					    BT_GAP_SCAN_FAST_INTERVAL)

int scan_init(void)
{
	int rc;

	rc = bt_le_scan_start(BT_LE_SCAN_PASSIVE_ALLOW_DUPLICATES, scan_device_found);
	if (rc) {
		LOG_ERR("Scanning failed to start (rc %d)", rc);
	}
	else {
		LOG_INF("Scanning Started");
	}

	return rc;
}
```

I'll try to get some new logs soon.  The issue is also present in these logs I took in the past: https://github.com/zephyrproject-rtos/zephyr/files/6716060/bt.zip

carlescufi, 2021-08-20:
@nwsetec Since it's a bit complex to reproduce the setup from our side, is there a chance you could retest with the latest main branch now that multiple fixes have been merged?

carlescufi, 2021-09-02:
@nwsetec do you have any additional information that we can use to try go figure out what the root cause is?

nwsetec, 2021-09-11:
@carlescufi I've tested the same application code with Zephyr SHA 30db452aec00b5e7f6f1933ef99a9a3387730ee2 and the issue is no longer present, so closing.

closed 2021-09-11 by nwsetec