zephyrproject-rtos/zephyr #36696: Json on native_posix_64 board

jdascenzio, 2021-07-02:
JSON don't work correctly on native_posix_64_board. 
I have create this descriptor:

```c
struct elt {
	int32_t  a;
	int32_t  b;
};

struct obj_array {
	struct elt elements[4];
	size_t num_elements;
};

static const struct json_obj_descr elt_descr[] = {
	JSON_OBJ_DESCR_PRIM(struct elt, a, JSON_TOK_NUMBER),
	JSON_OBJ_DESCR_PRIM(struct elt, b, JSON_TOK_NUMBER),
};

static const struct json_obj_descr obj_array_descr[] = {
	JSON_OBJ_DESCR_OBJ_ARRAY(struct obj_array, elements, 4, num_elements,
				 elt_descr, ARRAY_SIZE(elt_descr)),
};
```
Then, I encode a structure like this:
```c
	struct obj_array oa = {
		.elements = {
			[0] = { .a = 0,   .b = 1 },
			[1] = { .a = 2,   .b = 3 },
			[2] = { .a = 4,   .b = 5 },
			[3] = { .a = 6,   .b = 7 },
		},
		.num_elements = 4,
	};
	ret = json_obj_encode_buf(obj_array_descr, ARRAY_SIZE(obj_array_descr),
				  &oa, data,data_sz);
```
I have this result:
```json
{"elements":[{"a":0,"b":1},{"a":4,"b":5},{"a":4,"b":0},{"a":0,"b":0}]}
```
instead of:
```json
{"elements":[{"a":0,"b":1},{"a":2,"b":3},{"a":4,"b":5},{"a":6,"b":7}]}
```
I think there is a problem in the function bellow of json library. The alignment seems to don't work correctly
https://github.com/zephyrproject-rtos/zephyr/blob/6688f709964e0209162277bb2fcd178b774d318b/lib/os/json.c#L496

If someone could help me!
thx!

cfriedt, 2021-07-20:
@jdascenzio - are you able to submit a PR to fix?

cfriedt, 2021-08-07:
@jdascenzio - this works for me if I substitute `int64_t` for `int32_t`.

I believe the JSON library is designed to work with the host word size of whatever CPU it is running on.

On a 64-bit machine, it also fails for `int` although `int` is only required to be 32-bit on 64-bit machines.

To work both on 32-bit machines and 64-bit machines, I would suggest using `intptr_t` and `uintptr_t`. So 

```
struct elt {
	intptr_t  a;
	uintptr_t  b;
};
```

cfriedt, 2021-08-07:
Closing, as I believe this is probably not an actual bug.

commit 8757c71b (2023-06-29): json: Fix 64-bit support

This patch fixes encoding arrays of objects on 64-bit targets.

Fixes #36696
closed 2021-08-07 by cfriedt