zephyrproject-rtos/zephyr #36706: Some standard west commands should Just Work for native_posix board

palchak-google, 2021-07-02:
**Is your enhancement proposal related to a problem? Please describe.**
Currently when building for the `native_posix` board it is not possible to use a west command to execute the resulting binary. This is inconsistent with the workflow used for other boards. While not all west commands make sense for the `native_posix` board, it should still be possible to at least invoke the native binary somehow.

**Describe the solution you'd like**
If the target board is `native_posix` then:
* Running `west flash` should execute `${BUILD_DIR}/zephyr/zephyr.elf` with no arguments
* Running `west attach` should execute `${BUILD_DIR}/zephyr/zephyr.elf` with the argument `-attach_uart`

**Describe alternatives you've considered**
Alternatively, there should be a `west execute` command that is only enabled for the `native_posix` board.

**Additional context**

[ This space intentionally left blank ]

nashif, 2021-07-03:
this actually works with the arc simulator already, so I agree this could be made consistent...

```
i9:zephyr(main): west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner arc-nsim
Console now belongs to UART, hit CRTL-] to return to simulator.
*** Booting Zephyr OS build zephyr-v2.6.0-694-g3221a1e8e703  ***
Hello World! nsim
```

@palchak-google all of this can be done in cmake/west extensions in the tree, so patches welcome...

kartben, 2023-12-14:
@aescolar thoughts on this old issue? I wonder if it is still relevant in the light of native_sim

PR #68835 (2024-03-12): support 'west flash' with native_sim, add debugserver support
- west: add native_sim flash command
- west: runner: native_sim: add debugserver support

Fixes #36706