zephyrproject-rtos/zephyr #36724: The road to a stable Controller Area Network driver API

henrikbrixandersen, 2021-07-05:
The Controller Area Network driver API was introduced in Zephyr v1.14 but it is still marked as experimental here two years after.

This issue is intended to track the changes and actions required to get to the point where we can declare the CAN driver API as stable:

- [ ] Mark the CAN driver API as unstable to get the ball rolling (#36723)
- [ ] Solve the priority inversion issue (#26541)
- [ ] Unify the TX behavior of all CAN drivers (#19502)
- [ ] Announce state changes when the warning limit (96) is reached (#21010)

kentindell, 2021-07-09:
I'd just like to point out that eliminating priority inversion isn't an "enhancement": priority inversion is a dangerous flaw in a real-time system. When it is in CAN drivers, the latency of an urgent message can grow hugely from a few hundred microsecond to hundreds of milliseconds, and it can do this unpredictably, leading to Heisenbugs that cannot be replicated in a test environment.

This has already happened - on Mars. And for systems that are intended to be used with safety critical applications it must be fixed, even if that comes at the expense of average case performance.

henrikbrixandersen, 2021-07-09:
> I'd just like to point out that eliminating priority inversion isn't an "enhancement"

You are quite right. I have removed the enhancement label from this meta issue. I agree that this bug needs fixing.

alexanderwachter, 2021-07-11:
@kentindell you are right. We are working hard on this topic. The API change will include a rework of the send API call that makes it context based without any limits of buffers. Not sure about the ETA tough.
Thanks again for pointing out this bug! For me it was not obvious that we have a problem there.

closed 2023-01-10 by carlescufi