zephyrproject-rtos/zephyr #36734: osSignalWait crashes when called from a thread created with osCreateThread

jcpince, 2021-07-05:
**Describe the bug**
Hi, I was playing the the signals in cmsis on nrf5340dk and got a crash in osSignalWait or osSignalSet.

So, I ran the cmsis unit tests and the test around cmsis signals (test_signal_events_signalled) ran just fine.

So, after some debugging, I found out that the custom_data of the thread I was playing with was NULL in my code but valid in the test. It appears the difference is that the test is signalling the threads from the thread created to the main thread which works because the main thread has its custom_data set during the osCreateThread call (**but osCreateThread sets the main thread's custom_data  or _current instead of the created thread's custom data...**).

I modified the code so that the thread created get its custom data set and added some code to set the main thread's custom data with a static osThreadDef_t that I manually crafted and I can then signal and wait from any thread.

**To Reproduce**
Look at the modified unit test attached which is signalling the Thread_1 and Thread_2 from the main thread.

Note that the fix is hardcoded in this sample code (look at _current->custom_data = xxx;).

**Expected behavior**
We should be able to signal for any to any thread.

**Impact**
Crash.

**Logs and console output**
No logs

**Environment (please complete the following information):**
Zephyr commit tag: v2.6.0-rc1-ncs1, v2.6.0-ncs1-rc3

**Additional context**
As I tested, I would say that the custom_data pointer should be set to the current thread def and not the last thread created as it is currently implemented.
I also would suggest to create a thread def for the main thread is not already the case so it has not to be crafted manually.
[test_cmsis.c.gz](https://github.com/zephyrproject-rtos/zephyr/files/6764206/test_cmsis.c.gz)

nashif, 2021-09-08:
@jcpince can you submit this as a PR please?

jcpince, 2021-09-08:
Not sure this is a good idea, I wrote a patch to fix the issue that was occurring to me but I'd like to know if sending signals between the main thread and a cmsis thread is expected...
If it is expected to be ok, my patch my serve as a basis; if not, you should probably symply fail on the CMSIS APIs when not dealing with a cmsis thread.

closed 2021-11-23 by github-actions[bot]