zephyrproject-rtos/zephyr #36736: kernel: SMP global lock (and therefore irq_lock) works incorrectly on SMP platforms

evgeniy-paltsev, 2021-07-05:
**Describe the bug**
While debugging #36652 I've bumped into a situation when two threads are executing code protected with `SMP global lock` on two CPUs simultaneously (which shouldn't be happening)

The SMP global lock is widely used in Zephyr via `irq_lock` abstraction layer on SMP systems.

The SMP global_lock implementation looks quite suspicious. We have two complimentary `z_smp_global_lock`/`z_smp_global_lock` functions (which is OK) and one outstanding `z_smp_release_global_lock` which is called from `do_swap`.

The `z_smp_release_global_lock` previously had complimentary `z_smp_reacquire_global_lock` function which reacquired (if it is required) SMP global_lock on the thread brought online. However it was removed during huge scheduler rewrite 1acd8c2996c3a1ae0691247deff8c32519307f17

This seems to be incorrect as after thread swap (and call `z_smp_release_global_lock`) and swap back we will get `global_lock_count` inconsistent with `global_lock` value - the `global_lock` is reset by `z_smp_release_global_lock`, but thread's `global_lock_count` value is kept the same and all other calls to `z_smp_global_lock` and `z_smp_global_unlock1 by this thread won't work correctly.

This is one example of issue (among many possible ones):

```
CPU0                                                       CPU1
[1] lock - call z_smp_global_lock
[2] scheduled out, z_smp_release_global_lock call          lock - call z_smp_global_lock
                                                           we take global_lock here as it was
							   unlocked by z_smp_release_global_lock
							   on CPU0
[3] scheduled in
[4] unlock - call z_smp_global_unlock
    **Ooops!!!** - here we actually unlock global_lock
    taken by another CPU (CPU1)
```

This looks like a bug introduced by 1acd8c2996c3a1ae0691247deff8c32519307f17.

Here is the current implementation of SMP global_lock code [kernel/smp.c](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.6.0/kernel/smp.c)
```
static atomic_t global_lock;

unsigned int z_smp_global_lock(void)
{
	unsigned int key = arch_irq_lock();

	if (!_current->base.global_lock_count) {
		while (!atomic_cas(&global_lock, 0, 1)) {
		}
	}

	_current->base.global_lock_count++;

	return key;
}

void z_smp_global_unlock(unsigned int key)
{
	if (_current->base.global_lock_count) {
		_current->base.global_lock_count--;

		if (!_current->base.global_lock_count) {
			atomic_clear(&global_lock);
		}
	}

	arch_irq_unlock(key);
}

/* Called from within z_swap(), so assumes lock already held */
void z_smp_release_global_lock(struct k_thread *thread)
{
	if (!thread->base.global_lock_count) {
		atomic_clear(&global_lock);
	}
}
```

**Impact**
The SMP kernel works incorrectly which may lead to hangs (i.e #36652) / incorrect execution / data loss / etc.

evgeniy-paltsev, 2021-07-05:
CC kernel codeowners and kernel/sched.c, kernel/smp.c contributors
 @dcpleung @nashif @andyross @jharris-intel @pabigot @enjiamai @ceolin

andyross, 2021-07-06:
This is an intentional misfeature, actually.  Historically, code holding an irq_lock() was legally allowed to swap (despite the fact that the whole idea is nonsensical -- code above you that took the lock doesn't expect it to be magically released).  And this worked and was done in a few places in the kernel.  So this implements the same policy, which was IIRC required to get some tests to run at the time.  If you swap, you give up the lock and reacquire it on entry.

Note that spinlocks **don't** have this misfeature.  Please use those, and not the legacy irq_lock() API.

evgeniy-paltsev, 2021-07-07:
> If you swap, you give up the lock and reacquire it on entry.

@andyross But the issue is that we **don't** reacquire it on entry! The reacquiring was implemented initially but it was removed in 1acd8c2996c3a1ae0691247deff8c32519307f17 and now we don't even have a function which implements lock reacquire.

andyross, 2021-07-07:
Can you provide an exerciser?

At this point, several years after the transition, I'm personally more inclined to just eliminate this and say that the irq_lock() emulation layer isn't present and that you MUST use spinlocks on any SMP system.  Certainly nothing in the core system cares, this exists only for the benefit of porting code from uniprocessor setups that, again, should have been using spinlocks for the last few years anyway.

andyross, 2021-07-07:
Also, dropping priority.  The only way to hit this situation is when you are **swapping with an interrupt lock held**, which means you've already broken any reasonable synchronization scheme.  If you use irq_un/lock() to bound a proper critical section, this can't possibly happen.  It's only about the support for the legacy weirdness in the API that we fixed long ago.

evgeniy-paltsev, 2021-07-07:
> Can you provide an exerciser?

Do you mean to provide test case where issue happens or to provide the description about how the missing lock reacquire after swap may break the code?

> Also, dropping priority. The only way to hit this situation is when you are swapping with an interrupt lock held, which means you've already broken any reasonable synchronization scheme. If you use [...]

I'd disagree here as this mechanism is actually used in Zephyr generic code (i.e in posix thread layer). Zephyr user don't need to do something strange with irq_lock API to trigger this issue - it's enough to use some Zephyr functionality (like pthread) on SMP system which looks perfectly correct from user point of view.

So it should be priority: medium at least :)

andyross, 2021-07-07:
> I'd disagree here as this mechanism is actually used in Zephyr generic code (i.e in posix thread layer)

Can you post a separate bug about the code that is context switching with a lock held?  **THAT** we need to get fixed ASAP.  It really should just never happen, it was a quirk about the old API that we'd like to deprecate if possible.

evgeniy-paltsev, 2021-07-07:
> Can you post a separate bug about the code that is context switching with a lock held? THAT we need to get fixed ASAP. It really should just never happen, it was a quirk about the old API that we'd like to deprecate if possible.

Sure, I'll do that (at least for posix thread layer)

BTW, as I understand the `z_smp_release_global_lock` must be never called (as we must not use context switching with a lock held). So should we just call `z_fatal_error` in `z_smp_release_global_lock` instead of silently managing the global_lock which is done incorrect now?

andyross, 2021-07-07:
That would have the effect of breaking anyone who tries this trick, so... sure I guess?  But if that's the course we want, it makes more sense to just assert that global_lock == 0 in z_swap().

Really my aim here is to remove this API, not to fix it. :)

evgeniy-paltsev, 2021-07-07:
> That would have the effect of breaking anyone who tries this trick, so... sure I guess?

Yep, that's the aim. It would have saved hours of debugging and analyzing 200K+ instruction traces for me if it had been implemented :)

PR #37523 (2021-09-03): posix: fix pthread for SMP platforms
We shouldn't use swapping with an interrupt lock held as it works incorrectly on SMP platforms (see #36736 for details)

Fix that by replacing irq_lock (which is defined to SMP global lock on SMP systems) with spinlock for pthread subsystem.

NOTE: we fix that in a simple way with single spinlock for mutex / cond_var / barrier. That could be improved later (i.e. split it for several spinlocks).

Fixes: #36736
Fixes: #37109
closed 2021-09-03 by nashif