zephyrproject-rtos/zephyr #36745: Zephyr IP Stack Limited to 1514 bytes on the wire - no ICMPs beyond this limit

bob2oneil, 2021-07-05:
**Describe the bug**
There appears to be a fixed limit on the **MTU** size of the Zephyr IP stack. When I attempt some simply PING tests changing the size of the PING payload, or drive data into a virtual Ethernet interface (POSIX) where the external host created TAP interface has a very large MTU (e.g. 8k), and the data into the interface is verified with Wireshark to be fragment free, ICMP responses are generated by the stack only when the PING size is set to 1472 bytes, resulting in 1514 bytes on the wire. 

In this sense, it seems as if the **MTU** setting for each **iface** is somewhat meaningless and is capped to an internal limit.

This in on an interface that internal to Zephyr is set to an **MTU** size of 9K, with the host side of the TAP interface likewise set to 9K.

**What have you tried to diagnose or workaround this issue?**
Looked through source code for MTU limit or a kernel configuration to increase the stack's MTU.

**To Reproduce**
Steps to reproduce the behavior:
1. Standard build of an application.
2. west build -p always -b native_posix_64 ./

**Expected behavior**
Support for IP packets more in line with the MTU set for the Zephyr iface as well as the MTU set for the externally connected TAP interface.  So long as the network packets are not viewed as fragmented on a Wireshark capture, I would expect support for larger network packets, and in general perhaps ICMP responses for larger packet MTUs.

**Impact**
A problem since I want to pass IP jumboframes (9K) over one of the virtual interfaces.  Support for MTUs over 1500 are fairly typical.

**Logs and console output**

**Host PINGs at boundary 1472 or 1473:**
```

robocom@scnuc-1:~/RCSUS/bridge$ ping 138.0.0.1 -s 1472
PING 138.0.0.1 (138.0.0.1) 1472(1500) bytes of data.
1480 bytes from 138.0.0.1: icmp_seq=1 ttl=64 time=42.4 ms
1480 bytes from 138.0.0.1: icmp_seq=2 ttl=64 time=62.5 ms
1480 bytes from 138.0.0.1: icmp_seq=3 ttl=64 time=81.3 ms
1480 bytes from 138.0.0.1: icmp_seq=4 ttl=64 time=39.5 ms
1480 bytes from 138.0.0.1: icmp_seq=5 ttl=64 time=63.9 ms
^C
--- 138.0.0.1 ping statistics ---
6 packets transmitted, 5 received, 16.6667% packet loss, time 5007ms
rtt min/avg/max/mdev = 39.534/57.930/81.321/15.383 ms
robocom@scnuc-1:~/RCSUS/bridge$ ping 138.0.0.1 -s 1473
PING 138.0.0.1 (138.0.0.1) 1473(1501) bytes of data.

--- 138.0.0.1 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2055ms

```
**ifconfig on host:**

```
zeth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1600
        inet 192.0.1.2  netmask 255.255.255.0  broadcast 192.0.1.255
        inet6 fe80::200:5eff:fe00:53ff  prefixlen 64  scopeid 0x20<link>
        ether 00:00:5e:00:53:ff  txqueuelen 1000  (Ethernet)
        RX packets 4451  bytes 1094487 (1.0 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 5092  bytes 1276292 (1.2 MB)
        TX errors 0  dropped 347 overruns 0  carrier 0  collisions 0

zeth1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 9000
        inet 138.0.0.254  netmask 255.255.255.0  broadcast 138.0.0.255
        inet6 fe80::200:5eff:fe00:53ff  prefixlen 64  scopeid 0x20<link>
        ether 00:00:5e:00:53:ff  txqueuelen 1000  (Ethernet)
        RX packets 737  bytes 612286 (612.2 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2233  bytes 4470651 (4.4 MB)
        TX errors 0  dropped 292 overruns 0  carrier 0  collisions 0

```

**Shell output to command "**net iface**":**

```
[... 192 lines cut ...]
 - Booting Zephyr OS build zephyr-v2.6.0-734-gc1cff7558928

**Additional context**
Internal application logging using a raw socket indicates a packet decode as follows which indicates an internal cap on MTU at 1514 bytes, whereas Wireshark shows 1542 bytes on the wire.

```
ETHERNET RX:
 Interface: 'zeth1'
 Type: 'IPv4'(0x0800)
 Size: 1514
 Source MAC: 00:00:5e:00:53:ff
 Destination MAC: 02:00:5e:00:53:cc
   |-IP Version        : 4
   |-IP Header Length  : 20 DWORDS or 5 Bytes
   |-Type Of Service   : 0
   |-IP Total Length   : 1528 Bytes (Size of Packet)
   |-Identification    : 17656
   |-TTL               : 64
   |-Protocol          : 1
   |-Checksum          : 0xdb0d
   |-Source IP         : 138.0.0.254
   |-Destination IP    : 138.0.0.1
   |-Bridge Multicast  : false

```

jukkar, 2021-07-06:
The MTU in the device driver like Ethernet, is currently hardcoded in the driver when the device is created (search `NET_ETH_MTU` in the native_posix driver for example). So currently jumbo frames/packets are not supported.

bob2oneil, 2021-07-06:
Thanks Jukka for the quick response.

If I wanted to increase the MTU for say one virtual Ethernet interface, would it be possible for example, to simply change the source code to something like **eth_native_posix.c**, include this modified version as part of the Zephyr build - either in my _project_ scope as the definition of one virtual interface, or changing the Zephyr build itself to include this modified version, to increase the IP stack MTU to something around 9K?  

Changes might simply be to the arguments to the **ETH_NET_DEVICE_INIT** method as well as changing the runtime device MTU, and perhaps changes to the size of the _eth_context_ send and receive buffers.  Could this be done this way?

Or is this change much more pervasive and involves the entirety of the network stack, so that would be entirely futile?

Is there any consideration for having the MTU set at a kernel configuration parameter?  

```
struct eth_context {
	uint8_t recv[NET_ETH_MTU + ETH_HDR_LEN];
	uint8_t send[NET_ETH_MTU + ETH_HDR_LEN];
```

bob2oneil, 2021-08-04:
It is just fine if multiple POSIX interfaces need the same Jumboframe sized interface.  I would simply keep the MTU sized for one interest to around the 1514 limit.  But, I would want to drive the Jumboframe sized interface with the larger frames.  For Native-POSIX, the TAP device interface would be created externally and would be set to the 9K MTU size.  I am using both raw sockets and UDP sockets, but the read/writes to the raw socket interface using BSD sockets would need to support the Jumboframe size, as well presumably the network stack at the L2 level.  If the stack is designed for the more conventional size, then it will likely not work.  I would not be using any "real" device drivers for Native-POSIX.

rlubos, 2021-08-05:
Did you try the "temphack" from the post above? As I wrote, I was able to successfully ping Zephyr native posix "device" from my host with a jubo frame, merely with this change. As mentioned earlier, I don't see any obstacles in the IP stack itself that would prevent from handling large frames, but it would be good to give it a try with socket API as well. Just make sure to allocate enough net buffers to acoomodate all of the data.

bob2oneil, 2021-08-05:
Not yet, but I intend to do so and will close out this issue upon testing it.   Thanks.

bob2oneil, 2021-08-05:
Pings look good up to 8K MTU or 8042 on the wire.  I have not attempted to build a transmit buffer of this size for a raw packet transmit, but I think we are probably good.  I wrote a bash script to patch the source file for future Zephyr updates.  I am providing it here to the benefit of the community.  If there is a better way of handling changes to the source not bundled as part of the GIT process, please let me know.

```
#!/bin/bash
#
# Output framed banner status for major progress/milestone
separator="------------------------------------------------------------------------------------------"
outputBannerStatus() {
    echo ""
    echo $separator
    echo "$1"
    echo $separator
}

outputBannerStatus "Starting patch_zephyr_eth_native_posix script..."

# Script verions, update when changes are made
version="Thu 05 Aug 2021 05:23:31 PM EDT"

# Identify build and arguments to script
now="$(date)"
echo "Author: Bob O'Neil"
echo "Current date and time: $now"
echo "Script version: $version"
echo "Script arguments: ${@}"

# Location of patch file to increase size of MTU for the Native-POSIX Ethernet interface
FILE=$HOME/zephyrproject/zephyr/drivers/ethernet/eth_native_posix.c

# Backup file
BACKUPFILE=$FILE".bak"

echo "Verifying existence of Zephyr source file $FILE"
if [ -f "$FILE" ]; then
    echo "File found ..."
else
    echo "$FILE does not exist ..."
    exit 0
fi

echo "Checking for existence of backup file $BACKUPFILE"
if [ -f "$BACKUPFILE" ]; then
    echo "Backup file already exists"

    while true; do
        read -p "Do you wish to delete the backup file?" yn
        case $yn in
        [Yy]*) break ;;
        [Nn]*) exit ;;
        *) echo "Please answer yes or no." ;;
        esac
    done
fi

# Delete current backup file
if [ -f "$BACKUPFILE" ]; then
    rm $BACKUPFILE
fi

echo "Verifying existence of Zephyr source file $FILE"
if [ -f "$FILE" ]; then
    echo "File found ..."

    echo "Making backup file as $BACKUPFILE"
    cp $FILE $BACKUPFILE

    outputBannerStatus "Patching source file ..."
    
    # Use 'sed' stream editor to perform edits
    sed 's/struct eth_context {/#undef NET_ETH_MTU\n#define NET_ETH_MTU 9000\n\nstruct eth_context {/' $BACKUPFILE > $FILE
    
    echo "Patch successfully applied to increase Ethernet MTU"
fi

outputBannerStatus "File differences after applying patch"
diff $FILE $BACKUPFILE

now="$(date)"
outputBannerStatus "Exiting patch_zephyr_eth_native_posix script on $now"

```

closed 2021-08-05 by bob2oneil