zephyrproject-rtos/zephyr #36748: Zephyr IP Stack Leaks when using PROMISCOUS socket along with POSIX sockets based implementation.

bob2oneil, 2021-07-05:
**Describe the bug**
The Zephyr IP stack leaks and runs out of memory when using an interface **iface** configured for **promiscuous** behavior coupled with an application written using a standard POSIX implementation.

** Have you tried to diagnose or workaround this issue?**

Yes, looking for a way to dereference any cloned pkt in the stack, but the "struct net *pkt" corresponding to the POSIX socket's implementation is not available to dereference using the POSIX BSD sockets implementation, which simply provide pkt and packet length.

**To Reproduce**
Steps to reproduce the behavior:
1. west build application for POSIX target.
2. From command shell, verify interface is in promiscuous mode via "net iface"
3. Drive PING data with a larger MTU into the promiscuous interface.
4. Wait for the allocation failure manifesting itself as a 0 value for a packet buffer in the following method.

```
void net_pkt_cursor_init(struct net_pkt *pkt)
{
	pkt->cursor.buf = pkt->buffer;
	if (pkt->cursor.buf) {
		pkt->cursor.pos = pkt->cursor.buf->data;
	} else {
		pkt->cursor.pos = NULL;
	}
}
```

**Expected behavior**
Leak free behavior.

**Impact**
Showstopper for using a **promiscuous** interface coupled with POSIX BSD style sockets.

**Logs and console output**
Application log for the command "ping 138.0.0.1 -s 1472 -c 2" to limit the packet count for
diagnostics is as attached.

The callstack at the point of failure as as follows.  This is also shown in the Visual Studio screen capture.

_net_pkt_cursor_init(struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_pkt.c:1482)
net_if_recv_data(struct net_if * iface, struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_if.c:3865)
process_data(_Bool is_loopback, struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/include/net/net_pkt.h:280)
processing_data(struct net_pkt * pkt, _Bool is_loopback) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_core.c:141)
net_rx(struct net_if * iface, struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_core.c:361)
net_process_rx_packet(struct net_pkt * pkt) (/home/robocom/zephyrproject/zephyr/include/net/net_pkt.h:280)
tc_rx_handler(struct k_fifo * fifo) (/home/robocom/zephyrproject/zephyr/subsys/net/ip/net_tc.c:250)
z_thread_entry(k_thread_entry_t entry, void * p1, void * p2, void * p3) (/home/robocom/zephyrproject/zephyr/lib/os/thread_entry.c:29)
posix_thread_starter(void * arg) (/home/robocom/zephyrproject/zephyr/arch/posix/core/posix_core.c:305)
libpthread.so.0!start_thread(void * arg) (/build/glibc-eX1tMB/glibc-2.31/nptl/pthread_create.c:477)
libc.so.6!clone() (/build/glibc-eX1tMB/glibc-2.31/sysdeps/unix/sysv/linux/x86_64/clone.S:95)_

**Environment (please complete the following information):**
 - OS: (Ubuntu 20)
 - Toolchain (Zephyr SDK targeting Native POSIX board)
 - Commit SHA or Version used is Zephyr OS build zephyr-v2.6.0-734-gc1cff7558928

**Additional context**

Shell command "net iface" is as follows.  The 2nd interface as PROMISCOUS (138.0.0.1) is the one being tested:
```

Interface 0x555555602448 (Ethernet) [1]
===================================
Link addr : 02:00:5E:00:53:3B
MTU       : 1600
Flags     : AUTO_START,IPv4
Promiscuous mode : disabled
Ethernet capabilities supported:
	TXTIME
	Promiscuous mode
IPv4 unicast addresses (max 258):
[... 520 lines cut ...]
                    LOG_ERR("select() exception detected for wireless interface");
                }
            } // end wireless
        }

    } // end while

    LOG_INF("Stopping...");

    if (m_wiredInterface.isOpenSocket())
    {
        (void)zsock_close(m_wiredInterface.getSocket());
        m_wiredInterface.closeSocket();
    }

    if (m_wirelessInterface.isOpenSocket())
    {
        (void)zsock_close(m_wirelessInterface.getSocket());
        m_wirelessInterface.closeSocket();
    }
}
```
[image]

_[ZephyrPromiscuousSocketLeak.log](https://github.com/zephyrproject-rtos/zephyr/files/6766261/ZephyrPromiscuousSocketLeak.log)_

jukkar, 2021-07-06:
I am not seeing your app calling `net_promisc_mode_wait_data()` to receive the data. Did you see how `samples/net/promiscuous_mode` sample application is using the promiscuous API?

bob2oneil, 2021-07-06:
Thanks Jukka for your response.

This application uses standard BSD sockets to receive the data.

The entirely of the source is as attached.
[main.zip](https://github.com/zephyrproject-rtos/zephyr/files/6770145/main.zip)

rlubos, 2021-08-05:
@jukkar Please correct me if I'm wrong, but from what I see, the promiscous mode is implemented as a separate API and was not intended to work with socket API on top (see the aforementioned sample). As the promiscous mode API states that the application is responsible for releasing the packet after processing, it ends up in a memory leak in this case.

@bob2oneil  Given above, I don't think this issue should be considered as a "bug", as it was a design choice to make it work that way. If, from your perspective, promiscous mode should also work with the socket API, I'd rather see it as an enhancement to the stack. Feel free to opan a feature request issue for this, but please keep in mind that if noone else is interested in the feature you should be prepared to submit patches yourself.

bob2oneil, 2021-08-05:
The problem is that when dealing with the network API at the BSD socket level, there is no _net_pkt_ to release - just the raw data.  So if you want to have a **raw** L2 socket that is also **promiscuous**, so that the application can receive all packets on an interface (similar to what is done with PCap in Wireshark), the two can not be combined, that is a promiscuous raw socket.  

Since the BSD socket implementation above the net_if implementation obscures the underlying _net_pkt_ reference, there is nothing to free.  I suppose if there was some way of making the _net_pkt_ available to the BSD sockets layer, then the application could free it, but that would then change the POSIX API.  

Maybe the BSD sockets implementation could free up the _net_pkt_ reference **automatically** upon the read API as presumably the application now has the data?

I am fine with reclassifying this as an enhancement with the note that it just means Zephyr does not support **promiscuous raw sockets**, and that could be annotated somewhere.

rlubos, 2021-08-05:
> I am fine with reclassifying this as an enhancement with the note that it just means Zephyr does not support **promiscuous raw sockets**

As for the current implementation, it's a fair thing to say. The promiscuous mode is implemented as a separate API, and to my knowledge cannot be combined with socket API.

Current implementation of the promiscuous mode in Zephyr makes it work somehow separately from the rest of the IP stack. If promiscous mode is enabled on the interface, the low level processing will create a copy of the received packet (or reuse the orginal if it was otherwise dropped at L2), and puts it on a separate FIFO:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/net_if.c#L3867
This FIFO is only accessible by `net_promisc_mode_*` API, and not exposed to the rest of the stack in any way. Therefore making the promiscuous mode work with socket API is not a matter of a simple fix, but rather a complete rework of the existing promiscuous mechanism. That's why I think it rather falls under the "enhancement" umbrella.

bob2oneil, 2021-08-05:
I would argue that the only reason to have promiscuous mode is to have it available for reading via a socket API, so that you can combine promiscuous with a standard POSIX implementation.  But regardless, I am not advocating a ton of rework if it is a heavy lift.  In this case, I have a workaround, so whatever mechanism allows for reclassifying as an enhancement is fine.

bob2oneil, 2021-08-05:
It might be useful in the Zephyr documentation to mention features that while true by themselves, and not true when mixed with others. sort of like the danger of mixing alcohol with medications.  So for BSD sockets, it might be worthwhile to mention that they do not work with promiscuous settings.  There are other combinations of network features that work standalone, but not in other combinations that I have posted.

Thanks for all the effort Zephyr guys .. closing this baby.

closed 2021-08-05 by bob2oneil