zephyrproject-rtos/zephyr #36755: NTP client faults module when it fails

thedjnK, 2021-07-06:
**Describe the bug**
We are trying to use SNTP (NTP) to get an accurate representation of the time. This works OK when it gets it, but when it fails to get the time with -3 (ESRCH) the module faults

**To Reproduce**
Steps to reproduce the behavior:
Use sntp_simple() with an SNTP server (using time.windows.com), we are using ethernet.
When it is successful, the application runs fine, when it fails, the application faults:
```
[00:00:20.292,999] <wrn> ethernet_network: Get time from SNTP server failed! (-3)
[00:00:20.294,891] <err> os: ***** BUS FAULT *****
[00:00:20.294,891] <err> os:   Precise data bus error
[00:00:20.294,891] <err> os:   BFAR Address: 0x400000
[00:00:20.294,891] <err> os: r0/a1:  0xe000ed00  r1/a2:  0x00000002  r2/a3:  0x00400000
[00:00:20.294,891] <err> os: r3/a4:  0x410fd214 r12/ip:  0x20003760 r14/lr:  0x00055d3b
[00:00:20.294,921] <err> os:  xpsr:  0x01000000
[00:00:20.294,921] <err> os: Faulting instruction address (r15/pc): 0x0007537e
[00:00:20.294,921] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:20.294,921] <err> os: Current thread: 0x200066c8 (sysworkq)
```

**Expected behavior**
SNTP to work without faulting module if it fails

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: ARM GCC (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408
 - Commit SHA or Version used: Using nRF connect SDK 1.6 which uses zephyr 2.6

Testing on an nRF5340.

thedjnK, 2021-07-07:
Seems to be when NTP is used from a workqueue, have moved it to a separate thread and it's not faulting anymore

rlubos, 2021-07-07:
I'm afraid there's too little information for me to help. You seem to be using some custom application, would it be possible for you to provide a simple app that is able to reproduce the issue in a simulated environment preferably?

> Seems to be when NTP is used from a workqueue, have moved it to a separate thread and it's not faulting anymore

That could suggest there's some stack overflow issue, if the same code works fine from a different thread. Is your workqueue stack smaller from the other thread? Do you have a stack protection enabled? 

Also, the crash log indicates that you use system workqueue for the `sntp_simple()` call. That's no the best choice, given that `sntp_simple()` is a blocking call - once blocked every single module that depends on system workqueue is blocked. This could also be a possible issue, if some other module is not handling such large delays in processing time well...

closed 2021-09-20 by github-actions[bot]