zephyrproject-rtos/zephyr #36765: [PCI] ACRN sets Interrupt Line config space register to 0 and ReadOnly.

NanlinXie, 2021-07-07:
Environment

ACRN 2.4 Partition Mode / Zephyr guest and MSI (not MSI-X) capable physical device passed through to the OS.

HW/Board

EHL

Steps

Setup ACRN in partition mode and launch Zephyr as guest. PT a physical PCI device to the guest and assign an IRQ to it by using pcie_msi_vectors_allocate(), pcie_msi_vector_connect(), pcie_msi_enable() Zephyr API.
Expected result

Interrupt line register can be used by above mentioned functions to allocate a free IRQ line to the device and store the information later useage in sub-sequential called functions.

Actual result

Information which interrupt line is allocated to the device is lost as it cannot be stored to the interrupt line register. A later called isr vector connect function or irq enable function relys on a valid register value and maps or enables always a wrong isr/vector/interrupt line to the device.

Reproduce rate

Always

NanlinXie, 2021-07-07:
Linked to https://github.com/projectacrn/acrn-hypervisor/issues/6275

dcpleung, 2021-07-20:
@tbursztyka any thoughts?

jhedberg, 2021-07-21:
@dcpleung @tbursztyka is on vacation still this week, so we'll need to wait for his response. It's a bit unclear what exactly the "proper" behaviour should be, since when I look at how Linux deals with this it also looks like it's unconditionally writing into this register: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/pci/setup-irq.c#n61

lifeix, 2021-07-22:
@jhedberg Linux kernel is writing to it but it does not read it back. Even if the register is RW, it’s meant for INTx, OS is not supposed to use it for other purpose. 
For ACRN, we assume all PCI devices use MSI or MSI-X interrupt, so we don't support legacy INTx, thus the register is not implemented and wired to 0. In Linux kernel, it supports so many legacy devices, so the register is RW – but again, it’s not meant for the use case in Zephyr today.

jhedberg, 2021-07-22:
@lifeix not all drivers in Zephyr support MSI. E.g. the UART driver that's used for the console (`drivers/serial/uart_ns16550.c`) has no MSI support. That said, if I understood correctly, this is only an issue for drivers that do use MSI, i.e. where MSI-related code tries to read back the register value?

PR #37623 (2021-08-30): pcie: msi: pci_msi_enable() to take IRQ as parameter
This changes pci_msi_enable() to take IRQ number as a function
parameter. The old behavior relies on putting the IRQ number
into the interrupt line register in the PCI config space
during IRQ allocation, and reading it back when enabling IRQ.
However, the interrupt line register is only required to be
read-/writable when legacy interrupt is supported on the device.
Otherwise it has undefined behavior. On ACRN, they don't even
care about this register and always wires it to 0x00.
So this commit changes the behavior in pci_msi_enable() to not
 [... cut]
commit ec2b9d42 (2021-08-30): pcie: msi: pci_msi_enable() to take IRQ as parameter

This changes pci_msi_enable() to take IRQ number as a function
parameter. The old behavior relies on putting the IRQ number
into the interrupt line register in the PCI config space
during IRQ allocation, and reading it back when enabling IRQ.
However, the interrupt line register is only required to be
read-/writable when legacy interrupt is supported on the device.
Otherwise it has undefined behavior. On ACRN, they don't even
care about this register and always wires it to 0x00.
So this commit changes the behavior in pci_msi_enable() to not [... cut]
closed 2021-08-30 by jhedberg