zephyrproject-rtos/zephyr #36766: tests-ci :kernel.tickless.concept.tickless_slice : test failed

hakehuang, 2021-07-07:
**Describe the bug**
kernel.tickless.concept.tickless_slice test is failed on zephyr-v2.6.0-734-gc1cff7558928 on mimxrt1170_evk_cm7

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1170_evk_cm7  --sub-test kernel.tickless
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.6.0-734-gc1cff7558928  ***
Running test suite tickless_concept
===================================================================
START - test_tickless_sysclock
time 10, 8600
time 8610, 17200
PASS - test_tickless_sysclock in 4.267 seconds
===================================================================
START - test_tickless_slice
elapsed selapsed slice 4290, expected: <100, 110>
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/tickless/tickless_concept/src/main.c:58: thread_tslice: t <= SLICE_SIZE_LIMIT is false
elapsed slice 4300, expected: <100, 110>
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/tickless/tickless_concept/src/main.c:58: thread_tslice: t <= SLICE_SIZE_LIMIT is false
elapsed slice 0, expected: <100, 110>
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/tickless/tickless_concept/src/main.c:56: thread_tslice: t >= SLICE_SIZE is false
lice 8600, expected: <100, 110>
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/tickless/tickless_concept/src/main.c:58: thread_tslice: t <= SLICE_SIZE_LIMIT is false
FAIL - test_tickless_slice in 4.228 seconds
===================================================================
Test suite tickless_concept failed.
===================================================================
PROJECT EXECUTION FAILED

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v2.6.0-734-gc1cff7558928

hakehuang, 2021-07-07:
@mmahadevan108 is thickless support in 1170?

mmahadevan108, 2021-09-01:
It should be as this is the default setting in Zephyr

MaureenHelm, 2021-09-14:
I attempted to bisect but found that the test fails at the commit when the board was first introduced.

I suspect there is overflow problem due to the fast system clock (996 MHz). The test passes if I set `CONFIG_SYS_CLOCK_TICKS_PER_SEC=200`

```
$ west build -p auto -b mimxrt1170_evk_cm7 -T kernel.tickless.concept tests/kernel/tickless/tickless_concept -- -DCONFIG_SYS_CLOCK_TICKS_PER_SEC=200
```

danieldegrasse, 2022-01-20:
@dleach02 The RT1170 was switched over to the GPT as a clock source by #39449, so the issue with the frequency of systick shouldn't be present anymore. The test is passing on my RT1170 EVK now. I think we can close this issue if you are alright with that.

dleach02, 2022-01-20:
@hakehuang, can you confirm if this is still an open issue in your test range?

PR #83450 (2025-07-22): drivers: timer : cortex_m_systick MAX_TICKS protection
when CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC set to 960M and CONFIG_SYS_CLOCK_TICKS_PER_SEC set to 100
the MAX_TICKS will be zero or even a negative value, which is not expected, at this time the COUNTER_MAX is
so need add a protection here

fixes: #36766

there used to be a workaround, not a fix,

either change the CONFIG_SYS_CLOCK_TICKS_PER_SEC=200 or
CONFIG_PM to set the CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC to 32678
commit 8add9219 (2025-07-22): drivers: timer : cortex_m_systick MAX_TICKS protection

when CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC set to 960M
and CONFIG_SYS_CLOCK_TICKS_PER_SEC set to 100
the MAX_TICKS will be zero or even negative value, which is not
expected.
so need add a protection here downgrading the accuracy to
its as high as possible

also add build message to show that tickless has no effect

fixes: #36766

there used to be a workaround, not a fix,
either change the CONFIG_SYS_CLOCK_TICKS_PER_SEC=200
or
CONFIG_PM to set the CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC
to 32678
closed 2022-01-21 by hakehuang