zephyrproject-rtos/zephyr #36767: tests-ci :arch.arm.irq_advanced_features.arm_irq_target_state : test failed

hakehuang, 2021-07-07:
**Describe the bug**
arch.arm.irq_advanced_features.arm_irq_target_state test is failed on zephyr-v2.6.0-734-gc1cff7558928 on lpcxpresso55s28

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso55s28  --sub-test arch.arm
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.6.0-734-gc1cff7558928  ***
Running test suite arm_irq_advanced_features
===================================================================
START - test_arm_dynamic_direct_interrupts
PASS - test_arm_dynamic_direct_interrupts in 0.1 seconds
===================================================================
START - test_arm_zero_latency_irqs
Available IRQ line: 58
PASS - test_arm_zero_latency_irqs in 0.3 seconds
===================================================================
START - test_arm_irq_target_state
Available IRQ line: 57
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_irq_advanced_features/src/arm_irq_target_state.c:76: test_arm_irq_target_state: (result_state not equal to IRQ_TARGET_STATE_NON_SECURE)
Target state not set to Secure
FAIL - test_arm_irq_target_state in 0.23 seconds
===================================================================
Test suite arm_irq_advanced_features failed.
===================================================================
PROJECT EXECUTION FAILED

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v2.6.0-734-gc1cff7558928

hakehuang, 2021-07-07:
@dleach02  shall I disable this case for lpcxpresso55s28?

henrikbrixandersen, 2021-07-11:
Duplicate of #36416?

closed 2021-07-12 by hakehuang