zephyrproject-rtos/zephyr #36783: drivers: modem: hl7800

konovagit, 2021-07-07:
**Describe the bug**
When you call the function **hl7800_init** you have a loop to configure all gpios of the modem. In this loop you used **gpio_pin_configure** to configure gpio but when you need to configure pin as interrupt the bug occurred because an assert and **gpio_pin_configure** does not support interrupt configure (need to use: **gpio_pin_interrupt_configure**).

```
/* setup port devices and pin directions */
for (i = 0; i < MAX_MDM_CONTROL_PINS; i++) {
  ictx.gpio_port_dev[i] =
	  device_get_binding(pinconfig[i].dev_name);
  if (!ictx.gpio_port_dev[i]) {
	  LOG_ERR("gpio port (%s) not found!",
		  pinconfig[i].dev_name);
	  return -ENODEV;
  }
  
  ret = gpio_pin_configure(ictx.gpio_port_dev[i],
				  pinconfig[i].pin, pinconfig[i].config);
  if (ret) {
	  LOG_ERR("Error configuring io %s %d err: %d!",
		  pinconfig[i].dev_name, pinconfig[i].pin, ret);
	  return ret;
  }
}
```

When the loop index is on this pin (VGPIO), the flags GPIO_INT_EDGE_BOTH is not supported by **gpio_pin_configure** and cause assert.
```
/* MDM_VGPIO */
PINCONFIG(DT_INST_GPIO_LABEL(0, mdm_vgpio_gpios),
		DT_INST_GPIO_PIN(0, mdm_vgpio_gpios),
		(GPIO_INPUT | GPIO_INT_EDGE_BOTH)),
```

**To Reproduce**
Just need to compile and flash firmware with HL78 modem enables.

**Expected behavior**
All gpio will be configured without asserting.

**Impact**
Assert so the program can't start (modem init post-kernel).

**Fix**
BEFORE

```
ret = gpio_pin_configure(ictx.gpio_port_dev[i],
				pinconfig[i].pin, pinconfig[i].config);
```
AFTER FIX
```
if((pinconfig[i].config & GPIO_INT_ENABLE) == 0) {
	ret = gpio_pin_configure(ictx.gpio_port_dev[i],
				pinconfig[i].pin, pinconfig[i].config);
} else {
	ret = gpio_pin_interrupt_configure(ictx.gpio_port_dev[i],
				pinconfig[i].pin, pinconfig[i].config);
}
```

cfriedt, 2021-07-12:
@konovagit - could you possibly create a PR and reference this issue in your commit message with `Fixes #36783`.

Thanks!

rerickson1, 2021-09-23:
The proposed fixed above is not entirely correct. Need to fix this.

rerickson1, 2021-09-23:
Bumped this to medium to make sure to get this regression fixed and into the 2.7.0 release.

PR #38721 (2021-09-22): modem: hl7800: fix gpio with interrupts init
Need to use gpio_pin_interrupt_configure for any IO
that uses interrupts.

fix #36783
PR #38788 (2021-09-24): modem: hl7800: separate IO interrupt config
The interrupt config flags for an IO should be separate
from the standard IO configs because the interrupt config
is a separate API.

fixes #36783

Tested this on our modem. This fixes any ASSERTS and low power modes operate properly.
closed 2021-09-24 by nashif