zephyrproject-rtos/zephyr #36796: Build failure: samples/net/civetweb/http_server using target stm32h735g_disco

erwango, 2021-07-08:
**Describe the bug**
Build fail when compiling samples/net/civetweb/http_server using stm32h735g_disco target.
Issue only visible when explicitly running twister using this target.

@Benedikt Schmidt can you have a check ?

benediktibk, 2021-07-08:
Aye aye, captain

benediktibk, 2021-07-08:
Reproducible:
```
west build --board stm32h735g_disco -p auto samples/net/civetweb/http_server/  
...  
[187/194] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map 
: && ccache /opt/zephyr-sdk-0.12.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/home/bschmidt/Development/zephyr-playground/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/config/libsubsys__net__lib__config.a  zephyr/subsys/net/lib/dns/libsubsys__net__lib__dns.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  modules/civetweb/lib..__modules__lib__civetweb.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj  -L"/opt/zephyr-sdk-0.12.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/thumb/v7e-m/nofp"  -L/home/bschmidt/Development/zephyr-playground/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m7  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/bschmidt/Development/zephyr-playground/zephyr/build/zephyr && /usr/bin/cmake -E echo
/opt/zephyr-sdk-0.12.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `push_inner':
/home/bschmidt/Development/zephyr-playground/modules/lib/civetweb/src/civetweb.c:6267: undefined reference to `ferror'
/opt/zephyr-sdk-0.12.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `pull_inner':
/home/bschmidt/Development/zephyr-playground/modules/lib/civetweb/src/civetweb.c:6426: undefined reference to `fileno'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/bschmidt/Development/zephyr-playground/zephyr/build
```

galak, 2021-07-08:
Setting to high as we see this in the daily CI builds

benediktibk, 2021-07-08:
It builds for stm32h747i_disco_m7 and stm32h747i_disco_m4, which should be pretty close to stm32h735g_disco. I'll try to figure out where ferror and fileno are getting their implementations for these board definitions.

benediktibk, 2021-07-08:
Calls to ferror and fileno are both contained in the output of the preprocessor for civetweb.c, therefore I assume that they do not get thrown away by some defines and the problem must be during linking.

benediktibk, 2021-07-08:
modules/civetweb/lib..__modules__lib__civetweb.a seems to get compiled differently, depending on the board. In the bad case its relocation table references ferror, in the good case it does not.

benediktibk, 2021-07-08:
Created a pull request for the fix. Although I must say, I do not like the fix. It just relies upon optimizations to remove the calls to ferror/fileno. I think the better solution would be to check if these functions are available in civetweb, or implement them for the targets.

benediktibk, 2021-07-08:
PR: #36803

erwango, 2021-07-09:
Closing as https://github.com/zephyrproject-rtos/zephyr/pull/36803 is now merged. Txs @benediktibk !

closed 2021-07-12 by erwango