zephyrproject-rtos/zephyr #36802: MCUboot doesn't work with encrypted images on external flash

hjuul, 2021-07-08:
I have a custom board with stm32f777 and I use MCUboot with the secondary-slot in QSPI flash. Firmware updates work well as long as I don't turn on image encryption support.

But when I compile MCUboot with `-DCONFIG_BOOT_ENCRYPT_RSA=y` the image swap fails spectacularly, to the point where my device is bricked. This happens regardless of whether the new firmware image is encrypted or not.

**Logs and console output**
This is the MCUboot log when compiled with `-DCONFIG_MCUBOOT_LOG_LEVEL_DBG=y`:
```
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Swap type: test
D: erasing scratch area
D: initializing status; fa_id=3
D: writing swap_info; fa_id=3 off=0x3ffd8 (0x1fffd8), swap_type=0x2 image_num=0x0
D: writing swap_size; fa_id=3 off=0x3ffd0 (0x1fffd0)
D: writing enc_key; fa_id=3 off=0x3ffc0 (0x1fffc0)
D: writing enc_key; fa_id=3 off=0x3ffb0 (0x1fffb0)
D: writing magic; fa_id=3 off=0x3fff0 (0x1ffff0)
D: writing swap_info; fa_id=2 off=0x19ffd8 (0x1bffd8), swap_type=0x2 image_num=0x0
D: writing swap_size; fa_id=2 off=0x19ffd0 (0x1bffd0)
D: writing enc_key; fa_id=2 off=0x19ffc0 (0x1bffc0)
D: writing enc_key; fa_id=2 off=0x19ffb0 (0x1bffb0)
D: writing magic; fa_id=2 off=0x19fff0 (0x1bfff0)
D: erasing scratch area
D: erasing scratch area
D: erasing scratch area
D: erasing scratch area
E: 1: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
D: erasing scratch area
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
D: erasing scratch area
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
E: 2: Failed to send QSPI instruction
D: writing copy_done; fa_id=2 off=0x19ffe0 (0x1bffe0)
E: 2: Failed to send QSPI instruction
E: Unable to find bootable image
```
Then, when I cycle power a couple of times to try to recover I eventually get:
```
I: Primary image: magic=good, swap_type=0x4, copy_done=0x3, image_ok=0x3
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x20080000
E: r0/a1:  0x00000000  r1/a2:  0xfffa376c  r2/a3:  0x00000063
E: r3/a4:  0xffffffb3 r12/ip:  0x00000000 r14/lr:  0x080057db
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x08009e80
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20020838 (unknown)
E: Halting system
```

**Environment:**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.12.4
 - Zephyr v2.6.0
 - Board: Custom bord with stm32f777ni
 - QSPI flash: MT25QL256ABA

**Additional context**
I've traced the error messages down to the STM32 QSPI driver `flash_stm32_qspi.c`. The first error occurs on a QSPI READ command (0x03), for which the ST HAL returns `HAL_ERROR` (1). All other errors seem to occur because the ST HAL returns `HAL_BUSY` (2)

Laczen, 2021-07-08:
Hi @hjuul, could you share your dts definition file. I think that somehow the mcuboot region is being erased. It might be that your scratch area is on the qspi flash (which is unaccepatble for a encrypted setup).

hjuul, 2021-07-08:
This is the relevant part of my dts file. Scratch area is in internal flash.

```
&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(64)>;
			read-only;
		};
		assert_partition: partition@10000 {
			label = "assert-storage";
			reg = <0x00010000 DT_SIZE_K(64)>;
		};
		slot0_partition: partition@20000 {
			label = "image-0";
			reg = <0x00020000 DT_SIZE_K(1664)>;
		};
		scratch_partition: partition@1C0000 {
			label = "image-scratch";
			reg = <0x001C0000 DT_SIZE_K(256)>;
		};
	};
};

&quadspi {
	pinctrl-0 = <&quadspi_clk_pf10 &quadspi_bk1_ncs_pb6
		     &quadspi_bk1_io0_pd11 &quadspi_bk1_io1_pd12
		     &quadspi_bk1_io2_pf7 &quadspi_bk1_io3_pf6>;
	status = "okay";

	mt25ql256aba: qspi-nor-flash@0 {
		compatible = "st,stm32-qspi-nor";
		label = "MT25QL256ABA";
		reg = <0>;
		qspi-max-frequency = <54000000>;
		// 256 Megabits = 32 Megabytes 
		size = <DT_SIZE_M(32*8)>;
		status = "okay";

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			slot1_partition: partition@0 {
				label = "image-1";
				reg = <0x00000000 DT_SIZE_K(1664)>;
				};

			/* 0x1a0000 to 0x1FFFFF (384KiB) unused */
			factory_cal_partition: partition@200000 {
				label = "factory_cal";
				reg = <0x00200000 DT_SIZE_K(64)>;
			};
			/* 0x210000 to 0x2FFFFF (960KiB) unused */
			storage_partition: partition@300000 {
				label = "storage";
				reg = <0x00300000 DT_SIZE_M(61)>;
			};
		};
	};
```

Laczen, 2021-07-08:
@hjuul, OK, it's not the place of the scratch partition.

Could you add the logging output of a succesfull non-encrypted update. BTW are you sure it worked OK ? Did you use change the output somehow of updated the version number ?

Laczen, 2021-07-08:
@hjuul, there is something wrong in the size of the qspi flash device, this should be DT_SIZE_M(32). I don't think it is related, but you should adjust this as well as the storage area.

hjuul, 2021-07-12:
@Laczen Here's the output of the exact same setup, except I've set `CONFIG_BOOT_ENCRYPT_RSA=n` in MCUboot. It seems everything is working fine, except it takes a lot longer than expected (time seems to scale with the size of the defined partition, not the size of my image, ref [my recent Slack post](https://zephyrproject.slack.com/archives/CG77J8ZJS/p1625664553028900))
```
Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Swap type: test
D: erasing scratch area
D: initializing status; fa_id=3
D: writing swap_info; fa_id=3 off=0x3ffd8 (0x1fffd8), swap_type=0x2 image_num=0x0
D: writing swap_size; fa_id=3 off=0x3ffd0 (0x1fffd0)
D: writing magic; fa_id=3 off=0x3fff0 (0x1ffff0)
D: writing swap_info; fa_id=2 off=0x19ffd8 (0x1bffd8), swap_type=0x2 image_num=0x0
D: writing swap_size; fa_id=2 off=0x19ffd0 (0x1bffd0)
D: writing magic; fa_id=2 off=0x19fff0 (0x1bfff0)
D: erasing scratch area
D: erasing scratch area
D: erasing scratch area
D: erasing scratch area
D: erasing scratch area
D: erasing scratch area
D: writing copy_done; fa_id=2 off=0x19ffe0 (0x1bffe0)
I: Bootloader chainload address offset: 0x20000
I: Jumping to the first image slot
FIRMWARE HAS BEEN UPDATED
```
(the last line is the output of my application)

hjuul, 2021-07-12:
And @Laczen, thanks for the tip about the problem in my DTS file. As you expected it didn't fix this particular problem but very good to have it fixed anyway!

Laczen, 2021-07-12:
@hjuul, regarding the slow update (scaling with slot size), are you using confirmed images ?

Regarding the problem with encrypted images, have you checked that mcuboot is still smaller then 64kB ?

hjuul, 2021-07-12:
@Laczen Yes, I'm using confirmed images. Would that explain it?

It seems my bootloader is still smaller than 64KiB even with encryption enabled (this is with debug logging enabled):

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       58564 B        64 KB     89.36%
            SRAM:       48768 B       384 KB     12.40%
           SDRAM:          0 GB        32 MB      0.00%
            DTCM:          0 GB       128 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
```
With encryption disabled the image is about 10KB smaller.

Laczen, 2021-07-12:
> @Laczen Yes, I'm using confirmed images. Would that explain it?

When you are using confirmed images the image size becomes the slot size (something is written at the end of the image), and probably this also enforces mcuboot to copy all flash pages even if they contain 0xff. You could try and see if unconfirmed image do a faster swap.

> It seems my bootloader is still smaller than 64KiB even with encryption enabled (this is with debug logging enabled):
>
> ```
> Memory region         Used Size  Region Size  %age Used [...]

I am getting more and more convinced there is nothing wrong with the zephyr or configuration part. I guess there is something wrong with MCUboot support for RSA encryption (someone else has also reported problems with this).

I would propose closing this zephyr issue and creating a issue in the mcuboot github.

hjuul, 2021-07-20:
For reference, I wasn't using confirmed images after all, I was using unconfirmed _signed_ images. This behavior is surprising to me, but I agree it's probably not a Zephyr issue.

I'm closing this issue, but will continue the investigation and report any findings on the MCUboot github.

closed 2021-07-20 by hjuul