zephyrproject-rtos/zephyr #36808: xtensa xcc build  Fail ,   CONFIG_NO_OPTIMIZATIONS=y

hongshui3000, 2021-07-08:
**Describe the bug**
xtensa xcc build  Fail ,   
```
CONFIG_NO_OPTIMIZATIONS=y
```

**Logs and console output**
```
[144/151] Linking C executable zephyr\zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map 
cmd.exe /C "cd . && C:\usr\xtensa\XtDevTools\install\tools\RI-2021.6-win32\XtensaTools\bin\xt-xcc.exe   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr\zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=E:/work/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/arch/arch/xtensa/core/startup/libarch__xtensa__core__startup.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/xtensa/lib..sdk__soc__xtensa.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  modules/xtensa/libmodules_xtensa_hal.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -LE:/work/build/zephyr  -lgcc  -lhal  zephyr/arch/common/libisr_tables.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N && cmd.exe /C "cd /D E:\work\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E echo ""
app/libapp.a(main.c.obj):(.literal.z_log_msg2_runtime_vcreate+0x4): undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
app/libapp.a(main.c.obj): In function `z_log_msg2_runtime_vcreate':
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
zephyr/libzephyr.a(log_core.c.obj): In function `z_log_msg2_runtime_vcreate':
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
zephyr/libzephyr.a(usb_device.c.obj): In function `z_log_msg2_runtime_vcreate':
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
zephyr/libzephyr.a(usb_descriptor.c.obj): In function `z_log_msg2_runtime_vcreate':
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
zephyr/libzephyr.a(usb_transfer.c.obj):E:\work\zephyr\include\logging/log_msg2.h:57: more undefined references to `z_impl_z_log_msg2_runtime_vcreate' follow
zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a(xtensa-asm2.c.obj):(.literal.xtensa_excint1_c+0x1c): undefined reference to `z_log_minimal_printk'
zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a(xtensa-asm2.c.obj): In function `z_log_msg2_runtime_vcreate': 
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a(xtensa-asm2.c.obj): In function `xtensa_excint1_c':
E:\work\zephyr\arch\xtensa\core/xtensa-asm2.c:203: undefined reference to `z_log_minimal_printk'   
zephyr/soc/xtensa/lib..__soc__xtensa.a(soc.c.obj): In function `z_log_msg2_runtime_vcreate':
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
zephyr/kernel/libkernel.a(fatal.c.obj): In function `z_log_msg2_runtime_vcreate':
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
zephyr/kernel/libkernel.a(thread.c.obj): In function `z_log_msg2_runtime_vcreate':
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
zephyr/kernel/libkernel.a(mutex.c.obj): In function `z_log_msg2_runtime_vcreate':
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
zephyr/kernel/libkernel.a(mutex.c.obj): In function `adjust_owner_prio':
E:\work\zephyr\kernel/mutex.c:85: undefined reference to `z_log_minimal_printk'
zephyr/kernel/libkernel.a(mutex.c.obj): In function `z_impl_k_mutex_lock':
E:\work\zephyr\kernel/mutex.c:150: undefined reference to `z_log_minimal_printk'
zephyr/kernel/libkernel.a(mutex.c.obj): In function `z_impl_k_mutex_unlock':
E:\work\zephyr\kernel/mutex.c:247: undefined reference to `z_log_minimal_printk'
zephyr/kernel/libkernel.a(sched.c.obj): In function `z_log_msg2_runtime_vcreate':
E:\work\zephyr\include\logging/log_msg2.h:57: undefined reference to `z_impl_z_log_msg2_runtime_vcreate'
ninja: build stopped: subcommand failed.
```
**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain XCC

cfriedt, 2021-07-12:
Could you provide a bit more context for reproducibility?

E.g.
```
west build -p always -b $BOARD -t run  samples/subsys/logging/logger -DCONFIG_NO_OPTIMIZATIONS=y
```

Is this with the Zephyr SDK? If so, which revision?

hongshui3000, 2021-07-13:
@cfriedt 
The architecture I use is xtensa and the compiler is xcc. 
As long as CONFIG_NO_OPTIMIZATIONS=y, the program cannot be built, regardless of whether log is enabled or not.

nashif, 2021-10-20:
closing due to lack of response.