zephyrproject-rtos/zephyr #36819: qemu_leon3 samples/subsys/portability/cmsis_rtos_v2 samples failing

cgturner1, 2021-07-08:
When running twister on main encountered repeated failures for the qemu_leon3 arch for the following samples:

- samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher.semaphores
- samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher.same_prio
- samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher
- samples/subsys/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization

The failures all just show the boot banner and one line:

> *** Booting Zephyr OS build zephyr-v2.6.0-756-gb81cecf8dfc7  ***
>Wrote to message queue: 5

Was able to pinpoint when these samples started failing once commit 5d55730cf69dac7c8f99dcb17b04307cd050948c was introduced, before then there these samples pass as expected.

I am not sure if this has any impact on hardware boards as I have only ran this on qemu targets (note that this does not occur on any other qemu archs just leon3)

**To Reproduce**
Steps to reproduce the behavior:
1. Run twister for qemu_leon3 arch on the main branch for the samples above at any point past commit 5d55730cf69dac7c8f99dcb17b04307cd050948c

**Expected behavior**
Samples passing twister checks

**Impact**
A slight annoyance as this is a repeated failure on daily CI tests. Not sure if there is any real impact to physical hardware at this time.

**Logs and console output**
Console output for all tests:
> *** Booting Zephyr OS build zephyr-v2.6.0-756-gb81cecf8dfc7  ***
>Wrote to message queue: 5

Twister Logs for one of the samples
`DEBUG - Running cmake on /zephyrproject/zephyr/samples/subsys/portability/cmsis_rtos_v2/philosophers for qemu_leon3
DEBUG - Calling cmake with arguments: ['-B/zephyrproject/zephyr/twister-out/qemu_leon3/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher', '-S/home/cgturner/zephyrproject/zephyr/samples/subsys/portability/cmsis_rtos_v2/philosophers', '-DEXTRA_CFLAGS="-Werror"', '-DEXTRA_AFLAGS="-Wa,--fatal-warnings', '-DEXTRA_LDFLAGS="-Wl,--fatal-warnings"', '-DEXTRA_GEN_DEFINES_ARGS=--err-on-deprecated-properties', '-GUnix Makefiles', '-DDEBUG_PRINTF=1', '-DQEMU_PIPE=/zephyrproject/zephyr/twister-out/qemu_leon3/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher/qemu-fifo', '-DBOARD=qemu_leon3']
DEBUG - Finished building /zephyrproject/zephyr/samples/subsys/portability/cmsis_rtos_v2/philosophers for qemu_leon3
DEBUG - build test: qemu_leon3/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher
DEBUG - Building /zephyrproject/zephyr/samples/subsys/portability/cmsis_rtos_v2/philosophers for qemu_leon3
DEBUG - run test: qemu_leon3/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher
DEBUG - Spawning QEMUHandler Thread for qemu_leon3/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher
DEBUG - Running qemu_leon3/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher (qemu)
DEBUG - Spawning QEMUHandler Thread for qemu_leon3/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher
DEBUG - QEMU (63544): *** Booting Zephyr OS build zephyr-v2.6.0-806-gf818d8770b04  ***
DEBUG - QEMU (63544) complete (unexpected eof) after 0.12540984153747559 seconds
DEBUG - No timeout, return code from QEMU (63544): 2
DEBUG - return code from QEMU (63544): 2
DEBUG - run status: qemu_leon3/samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher failed
INFO - 1/1 qemu_leon3                samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher [31mFAILED [39mExited with 2 (qemu)`

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 LTS
 - Zephyr SDK
 - Latest Commit on main or at any point after commit Commit 5d55730cf69dac7c8f99dcb17b04307cd050948c

galak, 2021-07-13:
@martin-aberg

Seems like if memset is implemented as:

```
void *memset(void *buf, int c, size_t n)
{
        /* do byte-sized initialization until word-aligned or finished */

        unsigned char *d_byte = (unsigned char *)buf;
        unsigned char c_byte = (unsigned char)c;

        while (n > 0) {
                *(d_byte++) = c_byte;
                n--;
        }

        return buf;
}
```

that fails on qemu_leon3.

martin-aberg, 2021-07-14:
Thank you @cgturner1 and @galak for the report and analysis.
I can confirm that #36955 fixes the issue.

PR #36955 (2021-07-22): SPARC: Keep interrupts disabled during kernel init
This fixes an issue where interrupts could be taken after
`z_clock_driver_init()` while the system was still initializing. For
example when clearing large thread stacks during init.

Interrupts will now be enabled only when the first thread
starts or if `arch_irq_unlock()` is called.

Fixes #36819
closed 2021-07-22 by cfriedt