zephyrproject-rtos/zephyr #36849: drivers: serial: stm32: Software flow control

JonathanVanWin, 2021-07-09:
**Is your enhancement proposal related to a problem? Please describe.**
On the **stm32f411** errata sheet, at **2.5.5**, there is a hardware flow control problem
described (*nRTS signal abnormally driven low after a protocol violation*).

**Describe the solution you'd like**
I would like to add to the USART bindings an optional property for the **rts** pin.
The reason is that there is no way of knowing from the pinctrl-0 list, which pin is rts, because
the pinmux only contains: port, pin, and alternative function.
Then if hw-flow-control is enabled, only use CTS (if it is the CONFIG_SOC_STM32F411XE), and configure the
rts pin, and in async mode, when changing buffers set the pin high and lower it when finished.
The same when rx_disable occurs, and set it low again when enabled.

**Describe alternatives you've considered**
We could also instead of adding the rx_pin in the usart bindings, add pinctrl_names, as with the dma.

**Additional context**
Found out about this issue when testing the current async implementation, using **CONFIG_UART_ASYNC_API**
and running it on an stm32f411e_disco with hw-flow-control.
Had issues running it with flow control, and then found about the issue in the errata.

shlomow, 2021-07-19:
I think that this feature is very useful. Another use case is where the board has an error not having pin with rts alternate function, but it has some gpio. In this case, we have a workaround using the software flow control

average-potato28, 2026-08-25:
i am working on it.

closed 2026-09-08 by nashif