zephyrproject-rtos/zephyr #36851: FS logging backend assumes littlefs

protobits, 2021-07-09:
**Describe the bug**
FS Logging backend is assumed to be used with littlefs

**To Reproduce**
Steps to reproduce the behavior:
1. set: 
<pre>
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y

CONFIG_LOG_BACKEND_FS=y
</pre>
2. Got build error: `cannot find -lLITTLEFS`

**Expected behavior**
FS Logging backend should be usable with other filesystems.

**Impact**
Cannot use FAT FS

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: espressif
 - Commit SHA or Version used: v2.6.0

**Additional context**
Also, the help text on KConfig for CONFIG_LOG_BACKEND_FS explicitly mentions "littlefs"

protobits, 2021-07-10:
BTW, there's also this warning:
<pre>
/home/v01d/coding/embebit/ble_tracker/zephyr/subsys/logging/log_backend_fs.c:459:10: warning: initialization of 'void (*)(const struct log_backend * const)' from incompatible pointer type 'void (*)(void)' [-Wincompatible-pointer-types]
  .init = log_backend_fs_init,
</pre>

de-nordic, 2021-07-12:
I think that the logging system supports LittleFS. because, currently, this is the only file system that supports auto-mount.

cfriedt, 2021-07-12:
Might be necessary to add a `depends` in Kconfig and potentially a runtime check. Any thoughts, @de-nordic ?

de-nordic, 2021-07-12:
> 
> 
> Might be necessary to add a `depends` in Kconfig and potentially a runtime check. Any thoughts, @de-nordic ?

Yeah, I think it should be. There is dependency on `FILE_SYSTEM` instead, because the original work has been to let any FS.
The problem is that the fs needs to be initialized quite early, for that purpose, but the FAT FS uses different flash access than the Littlefs, and it was not possible to get both into auto-mount.

protobits, 2021-07-12:
I haven't yet confirmed it is possible to use FAT based logging to SD card as I'm having issues writing to SD card using ESP32 but my understanding was that I would mount the card at boot on my own. Also, looking at the code I understood that logging attempts to access the file based backend on every log operation and it would eventually succeed once the file is available.

de-nordic, 2021-07-12:
It probably is; the problem is that the logging will start when your application, or something else, explicitly mounts the FAT FS.

mrfuchs, 2021-08-18:
I can confirm that logging to a FAT filesystem works.

As mentioned above, I had to remove/comment the littlefs include first:

https://github.com/zephyrproject-rtos/zephyr/blob/d04a58bd1e6cedd5e6957c7e6149ea011d4aa698/subsys/logging/CMakeLists.txt#L29

I then enabled the FS logging backend and pointed its output to the flash volume mountpoint `/NAND`:

```
CONFIG_LOG_BACKEND_FS=y
CONFIG_LOG_BACKEND_FS_DIR="/NAND:"
...
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y

CONFIG_SPI=y
CONFIG_FLASH=y
CONFIG_SPI_NOR=y
CONFIG_SPI_NOR_CS_WAIT_DELAY=0
CONFIG_SPI_NOR_SFDP_DEVICETREE=y
CONFIG_SPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096
...
CONFIG_DISK_DRIVER_FLASH=y
CONFIG_DISK_FLASH_DEV_NAME="W25Q16JVSNIQ"
CONFIG_DISK_FLASH_START=0x8000
CONFIG_DISK_VOLUME_SIZE=0x1F0000
CONFIG_DISK_FLASH_MAX_RW_SIZE=4096
CONFIG_DISK_FLASH_ERASE_ALIGNMENT=0x1000
CONFIG_DISK_ERASE_BLOCK_SIZE=0x1000
CONFIG_DISK_FLASH_SECTOR_SIZE=4096
CONFIG_FS_FATFS_MAX_SS=4096
```

de-nordic, 2021-08-18:
> 
> I then enabled the FS logging backend and pointed its output to the flash volume mountpoint `/NAND`:
> 

Yes, but you had to manually mount the FAT device before getting any logs written, right?

mrfuchs, 2021-08-18:
> Yes, but you had to manually mount the FAT device before getting any logs written, right?

Yes, exactly.

nvlsianpu, 2021-08-18:
The backend allow to start without FS volume mounted.

de-nordic, 2021-08-18:
> 
> 
> The backend allow to start without FS volume mounted.

It is then possible to allow logging to FAT placed log file, only issue is that the user will be responsible for mounting the file system; if we add Kconfig for that, then it needs to be clearly stated that logging will fail unless system is mounted by an user.

PR #37860 (2021-09-01): logging: fs: Remove littlefs dependency
Remove littlefs dependency as FS backend works with any file system as long as
it is (manually or automatically) mounted.

Fixes #36851
commit 05a1a5c4 (2021-09-01): logging: fs: Remove littlefs dependency

Remove littlefs dependency as FS backend works with any file system as
long as it is (manually or automatically) mounted.

Fixes #36851
closed 2021-09-01 by nashif