zephyrproject-rtos/zephyr #36852: acrn_ehl_crb:  the test of tests/subsys/cpp/libcxx/ failed

enjiamai, 2021-07-09:
**Describe the bug**
The test case of tests/subsys/cpp/libcxx/ failed.  (cpp.libcxx.cpp.libcxx.newlib.exception)

After bisect, the failure start from this commit : 518dd32efee6ad999432327f316e3cf0feabe47b
"tests: cpp: libcxx: Use full version of newlib for testing"

Note: this issue only happened on acrn_ehl_crb board.

**To Reproduce**
Steps to reproduce the behavior:
1. Run this command from the zephyrproject:
twister -p acrn_ehl_crb --device-testing --device-serial-pty="/opt/remotehw/acrn-test-pty.exp,ehlsku11" --west-flash="/opt/remotehw/remotehw-x86-acrn.sh,ehlsku11" -vv -T tests/subsys/cpp/libcxx/

2.Check handler.log in twister.out in directory to see the error log.

**Expected behavior**
The test case is passed as expected.

**Impact**
Not sure.

**Logs and console output**
```
ACRN:\>vm_console 0
----- Entering VM 0 Shell -----
*** Booting Zephyr OS build zephyr-v2.6.0-intel-ww28.2  ***
version 201703
Running test suite libcxx_tests
===================================================================
START - test_array
PASS - test_array in 0.1 seconds
===================================================================
START - test_vector
PASS - test_vector in 0.1 seconds
===================================================================
START - test_make_unique
PASS - test_make_unique in 0.1 seconds
===================================================================
START - test_exception
exit

```

**Environment (please complete the following information):**
 - OS: Linux, 
 - Toolchain zephyr-sdk-0.12.4
 - Commit SHA or Version:  commit after 518dd32efee6ad999432327f316e3cf0feabe47b
    ACRN version: github v2.5-unstable (acrn-2021w06.5-180000p) commit: 90420123e

**Additional context**

stephanosio, 2021-07-12:
> Note: this issue only happened on acrn_ehl_crb board.

@enjiamai Is the same failure also observed on `acrn` board?

stephanosio, 2021-07-12:
@enjiamai Also you noted here that the failure starts from https://github.com/zephyrproject-rtos/zephyr/commit/518dd32efee6ad999432327f316e3cf0feabe47b.

Could you double check that it is actually from https://github.com/zephyrproject-rtos/zephyr/commit/518dd32efee6ad999432327f316e3cf0feabe47b and not https://github.com/zephyrproject-rtos/zephyr/commit/7e47ef281eccd7d1391bcdcf92affd7493c0dbd3?

enjiamai, 2021-07-12:
> > Note: this issue only happened on acrn_ehl_crb board.
> 
> @enjiamai Is the same failure also observed on `acrn` board?

Hi @stephanosio , the is also failed on acrn board

enjiamai, 2021-07-12:
> Could you double check that it is actually from [518dd32](https://github.com/zephyrproject-rtos/zephyr/commit/518dd32efee6ad999432327f316e3cf0feabe47b) and not [7e47ef2](https://github.com/zephyrproject-rtos/zephyr/commit/7e47ef281eccd7d1391bcdcf92affd7493c0dbd3)?

Sorry, @stephanosio , after I double check it, the failed commit is start from 615371900ed4165fefa5c03f7b6df34ac18bdefa
bad  => 615371900ed4165fefa5c03f7b6df34ac18bdefa (tests: cpp: libcxx: Test with full and nano variants of newlib)
good => 518dd32efee6ad999432327f316e3cf0feabe47b (tests: cpp: libcxx: Use full version of newlib for testing)
good => f0e3280c61e6794038e01dcadc167943e8f1b990 (cpp: Fix C++ exception handling newlib dependency)
good => 7e47ef281eccd7d1391bcdcf92affd7493c0dbd3 (linker: cpp: Fix C++ exception handling info section linking)

yerabolu, 2021-10-01:
cpp.libcxx.cpp.libcxx.newlib.exception
cpp.libcxx.cpp.libcxx.newlib_nano.array
cpp.libcxx.cpp.libcxx.newlib_nano.vector
cpp.libcxx.cpp.libcxx.newlib_nano.make_unique
cpp.libcxx.cpp.libcxx.newlib_nano.exception

Tests are still failing on commit: 5130db30058c

enjiamai, 2021-10-18:
Hi @stephanosio , do you have any idea or suggestions about this issue? thanks!

commit 8188fd3b (2021-11-08): tests: subsys: libcxx: extend the timeout for cpp.libcxx.newlib_nano

Extend the timeout to prevent the cpp.libcxx.newlib_nano testcase runs
failed in acrn_ehl_crb, ehl_crb and up_squared. We need to give these
boards more time to finish the testcase's automation.

Fixes #36852