zephyrproject-rtos/zephyr #36857: i2c_samd0.c burst write not compatible with ssd1306.c

Tubbles, 2021-07-09:
Running Zephyr with a SSD1306 compatible OLED screen does not work over I2C

I am running with a [seeeduino xiao](https://wiki.seeedstudio.com/Seeeduino-XIAO/) and its accompanying [dev kit](https://wiki.seeedstudio.com/Seeeduino-XIAO-Expansion-Board/) which features a SSD1306 compatible 128x64 pixels OLED screen, accessed over the I2C bus.

Comparing with a similar library from the Arduino ecosystem I saw that the I2C packets of Zephyr was incorrectly chopped up into smaller chunks. I tracked it down and it seems that it comes from the usage of `i2c_burst_write()`, but I don't know what is the supposed behavior of the I2C driver. Is it okay for it to inject STOP conditions between each I2C message? There is a warning in the header file which seems to indicate that it is.

**To Reproduce**
1. checkout the below zephyr version
2. apply the changes in the below attached patch (except the one for the ssd1306.c file which contains a workaround)
3. unpack blinken.zip below
4. `west build -b seeeduino_xiao`
5. `west flash`
6. See error

**Expected behavior**
A stripey pattern on the display

**Impact**
Showstopper for any application using this driver combo (and possibly others with the ssd1306?)

**Logs and console output**
Expected comms:
[image]

Actual comms:
[image]

**Environment (please complete the following information):**
 - Manjaro Linux
 - zephyr-v2.6.0-740-gc32a96af6ade

**Additional context**
[0001-seeduino-xiao-add-i2c-support-fix-ssd1306-burst-writ.patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/6794058/0001-seeduino-xiao-add-i2c-support-fix-ssd1306-burst-writ.patch.txt)
[blinken.zip](https://github.com/zephyrproject-rtos/zephyr/files/6794100/blinken.zip)

jfischer-no, 2021-07-14:
This is not a ssd1306 controller driver issue but i2c_samd0.c driver bug/incompatibility.

Tubbles, 2021-07-15:
> This is not a ssd1306 controller driver issue but i2c_samd0.c driver bug/incompatibility.

Is there some more specification anywhere on how the API implementer is supposed to handle this situation? If it is a bug, we need to fix the driver; on the other hand, if its an incompatibility, we need to change what API we are using..

carlescufi, 2021-07-20:
@Tubbles there is a lengthy discussion regarding i2c burst writes in [this PR](https://github.com/zephyrproject-rtos/zephyr/pull/34066). If your issue is similar to this maybe you want to weigh in there.

rustypig91, 2021-09-04:
> This is not a ssd1306 controller driver issue but i2c_samd0.c driver bug/incompatibility.
Stumbled across the same issue with ssd1306 and seeeduino xiao
This is my fix https://github.com/rustypig91/zephyr/commit/2b98cfb2bcb98662736e4ed35bdf75314e763bc1

dstahlke, 2021-09-12:
@rustypig91 This fix works for me, for both the Adafruit OLED FeatherWing (SH1107) and the common 0.96" OLED breakout board (SSD1306?).  Very nice work - it took way too much time to debug this to the point where I found this ticket.  Could you submit a pull request for this?

dstahlke, 2021-09-12:
i2c_burst_write does have a warning saying it's not portable and doesn't work correctly on some devices.  It'd be nice if ssd1306.c could use i2c_transfer instead.

Looking through the code, it appears this would be easy to do for all the command buffers (just prepend the command byte), but for ssd1306_write would not be so easy because the data comes from a user-supplied buffer.

dstahlke, 2021-09-12:
Relevant issues: #20154 #33606

Tubbles, 2021-09-12:
> This fix works for me, for both the Adafruit OLED FeatherWing (SH1107) and the common 0.96" OLED breakout board (SSD1306?).

What processor are you using?

dstahlke, 2021-09-12:
@Tubbles ATSAMD21G18, on the Adafruit Feather M0 Basic board.

rustypig91, 2021-09-12:
#38482 created

PR #38482 (2021-09-27): drivers: i2c: sam0: Continue write/read if next message allows it
Update sam0 i2c driver to directly send/receive next message if it is
in the same direction and the current message has no stop or restart
flags. Seems like in some drivers this is the expected behaviour.

Fixes #36857
commit 2ec4986d (2021-09-27): drivers: i2c: sam0: Continue write/read if next message allows it

Update sam0 i2c driver to directly send/receive next message if it is
in the same direction and the current message has no stop or restart
flags. Seems like in some drivers this is the expected behaviour.

Fixes #36857
closed 2021-09-27 by cfriedt