zephyrproject-rtos/zephyr #36859: Possible Advertising PDU Corruption if bt_enable called in SYS_INIT function

legoabram, 2021-07-10:
**Describe the bug**
I've encountered an extremely bizarre behavior that occasionally occurs if `bt_enable` is called within a function called using the `SYS_INIT` macro, and using a connectable advertising set. The first transmission made in this configuration doesn't complete correct. The following graph shows GPIO lines that I've tied to the `NRF_RADIO_EVENT_TXREADY`, `NRF_RADIO_EVENT_RXREADY`, `NRF_RADIO_EVENT_ADDRESS`, radio events and the PA/LNA PPI fork channels.

[image]

It looks like the RADIO interrupt isn't processed after the first TX is complete, since the PA/LNA haven't been setup for the RX. The glitch is fixed when the next advertisement is started, but during this time PACKETPTR is still pointing to the advertisement PDU. As a result, if another advertisement from another device occurs within this time frame, it will override the PDU in memory. This also results in the advertisement then mimicking the other device, which can be seen in these screenshots from nRF Connect. In my example, two advertising sets are used. The yellow box is correct, and the green box is the mimicked device. The detailed view on the right is for the mimicked device.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Zephyr application using the provided `main.c` and `prj.conf` [prj.zip](https://github.com/zephyrproject-rtos/zephyr/files/6794526/prj.zip)
2. mkdir build; cd build
3. cmake -DBOARD=board\nrf52840dk_nrf52840
4. make
5. Ensure that another nearby device is advertising at a high interval. The error may not show on every run.
6. Using nRF Connect, you will only see one advertising set from the target device when the error occurs.

**Expected behavior**
When working correctly, the `(iBeacon)` for the target device will switch on and off, and different advertising packets will show in the details.

**Impact**
I've been able to mitigate the issue by changing when bt_enable is called.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: GNU Arm Embedded 10 2020-q4-major
 - Commit SHA or Version used db975604

cvinayak, 2021-07-11:
I am not sure if bt_enable is callable from SYS_INIT with "any" (priority) order of initialization of modules in Zephyr application. Other Bluetooth Subsystem threads are needed to be executed in co-operation to ensure proper operation of the stack.

Setting the priority of this issue low, consider that `bt_enable` is called from system initialization there will be high interrupt latencies introduced causing Bluetooth Controller to fail.

carlescufi, 2021-07-15:
@legoabram `bt_enable()` was designed to be called from `main()` or a user thread, not `SYS_INIT`. There could be several reasons why this is causing issues, including high interrupt activity during the init phase or an unitialized driver required by Bluetooth. I suggest you move the call to the main execution context instead.

carlescufi, 2021-07-29:
Please call `bt_enable()` from `main()` and not `SYS_INIT`. If the problem persists once you've made that change, please reopen the issue.

closed 2021-07-29 by carlescufi