zephyrproject-rtos/zephyr #36865: k_work_q seems to check uninitialized flag on start

jefffhaynes, 2021-07-10:
When initializing a work queue with `k_work_q_start`, the passed work_q stuct is not initialized (nor do any of the documentation or examples indicate that it should be).  Consequently, if the `K_WORK_QUEUE_STARTED_BIT` happens to be set via dumb luck, this assert will fail:

https://github.com/zephyrproject-rtos/zephyr/blob/db9756045e1ae2282ac5c23c98e81d77b35ad19b/kernel/work.c#L674

galak, 2021-07-13:
The question is what behavior one expects if we call this API with an already init kwork queue.

I'm guessing the assert was an attempt to check for that case.

jefffhaynes, 2021-07-13:
Sure, I understand that but the issue is that this creates a sort of magic line that you can cross without warning and then you're in an assert condition.  It means this method can only be called "so often", where "so often" is largely an unknowable value.  I would consider either deprecating this in favor of the callback version or returning an error code on failure rather than bringing down the system.