zephyrproject-rtos/zephyr #36866: CONFIG_NO_OPTIMIZATIONS=y MPU fault on Zephyr 2.6

1am, 2021-07-10:
**Describe the bug**

Hello. I've been trying to move a small project built around Zephyr 2.5 to v2.6 and kept getting one issue which I couldn't figure out -  Whenever I've set `CONFIG_NO_OPTIMIZATIONS=y` (for debugging) my application fails. When I set `CONFIG_NO_OPTIMIZATIONS=n` the code works. I'm using nrf52832DK as the target for this test.

Eventually setting my app aside I went to work with https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v2.6.0/samples/bluetooth/scan_adv and can also repeat a crash with `CONFIG_NO_OPTIMIZATIONS=y`.
The only solution I've found so far https://github.com/zephyrproject-rtos/zephyr/issues/19340#issuecomment-723876554 doesn't work in my case. The error is the following:

```
00> *** Booting Zephyr OS build zephyr-v2.6.0  ***
00> 
00> Starting Scanner/Advertiser Demo
00> 
00> [00:00:00.334,350] <err> os: ***** MPU FAULT *****
00> [00:00:00.334,381] <err> os:   Stacking error (context area might be not valid)
00> [00:00:00.334,381] <err> os:   Data Access Violation
00> [00:00:00.334,411] <err> os:   MMFAR Address: 0x2000283c
00> [00:00:00.334,411] <err> os: r0/a1:  0x20000888  r1/a2:  0x00000001  r2/a3:  0x20000888
00> [00:00:00.334,442] <err> os: r3/a4:  0x20000888 r12/ip:  0x20002830 r14/lr:  0x200003d8
00> [00:00:00.334,442] <err> os:  xpsr:  0x20000200
00> [00:00:00.334,472] <err> os: Faulting instruction address (r15/pc): 0x20000620
00> [00:00:00.334,472] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
00> [00:00:00.334,503] <err> os: Current thread: 0x20000668 (unknown)
00> [00:00:01.117,126] <err> os: Halting system
```

Surprisingly the scan_adv from v2.6 and v2.5 fails the same way with `CONFIG_NO_OPTIMIZATIONS=y` though my application worked well with  Zephyr 2.5 and `CONFIG_NO_OPTIMIZATIONS=y` - Not sure how and why.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v2.6.0/samples/bluetooth/scan_adv
2. Add these to prj.conf
```
CONFIG_IDLE_STACK_SIZE=340  # from https://github.com/zephyrproject-rtos/zephyr/issues/19340#issuecomment-723876554
CONFIG_NO_OPTIMIZATIONS=y  # Toggle this for MPU fault vs ok
CONFIG_DEBUG=y
CONFIG_CONSOLE=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
```
3. `west build --pristine -b nrf52dk_nrf52832 . && west flash`
4. Get the error on RTT
5. Set `CONFIG_NO_OPTIMIZATIONS=n`
6. It will work ok

**Expected behavior**

Expected that the build would be bigger but it will work.

**Impact**

Nobody will get hurt or anything but it would be really cool to be able to debug non optimized code. Had big plans on working with Zephyr and already invested some time.

**Environment (please complete the following information):**
- Linux
- Zephyr 2.5.0, 2.6.0

alexanderwachter, 2021-07-11:
You are running into a stack overflow, because your code needs more stack when it is not optimized.

1am, 2021-07-11:
Thank you @alexanderwachter. Indeed setting `CONFIG_BT_RX_STACK_SIZE=2048` helps though in case of my app I'm now getting another issue.

```
00> [00:00:01.350,830] <err> os: ***** USAGE FAULT *****
00> [00:00:01.350,860] <err> os:   Attempt to execute undefined instruction
00> [00:00:01.350,891] <err> os: r0/a1:  0x00000001  r1/a2:  0x0000000d  r2/a3:  0x0000000d
```

Since it's a different problem I'll close this issue and try to figure it out.

closed 2021-07-11 by 1am