zephyrproject-rtos/zephyr #36869: Direction Finding Connectionless porting to nrf52811

Faizan-Ellahi-DE, 2021-07-11:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
Describe in details the hardware support being requested and why this support benefits Zephyr.

**Describe why you are asking for this support?**
Describe why you are asking for this support instead of contributing it directly to the tree

If this is a new board or SoC, please state whether you are willing to maintain the Zephyr support for it if it is included in the main tree

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the hardware here.

Faizan-Ellahi-DE, 2021-07-11:
Hello,

Faizan-Ellahi-DE, 2021-07-11:
Hello,
I am looking to port the existing code "Direction Finding Connection TX" to nrf52811, I am stuck at bt_df_set_adv_cte_tx_param(adv_set, &cte_params) function, it returns an error -134 (Unsupported Value), after some deep diving i have found that my code is not going inside le_df_init() function because BT_FEAT_LE_CONNECTIONLESS_CTE_TX(bt_dev.le.features) is not returning a non-zero value. Kindly guide me what i am missing.
File location : zephyrproject\zephyr\subsys\bluetooth\host\hci_core.c
[image]

Faizan-Ellahi-DE, 2021-07-12:
File Location: \zephyr\soc\arm\nordic_nrf\nrf52\Kconfig.soc
Sdd Radio BLE Direction Finding Flag in NRF52811 soc config

closed 2021-07-12 by Faizan-Ellahi-DE

Bosemani, 2022-01-24:
Hai @Faizan-Ellahi-DE I didn't get your point
I'm also using nrf52811 and getting the same error.
can you explain in detail pls?