zephyrproject-rtos/zephyr #36872: Integrate Android bluedroid SBC decoder/encoder

MarkWangChinese, 2021-07-12:
## Origin

Android
https://cs.android.com/android/platform/superproject/+/master:system/bt/embdrv/sbc/

## Purpose

Implement the SBC decoder and encoder.

## Mode of integration

It is Apache 2.0 license, and it is maintained by the Android.
Suggest to integrate this as one external component, the repository name as "sbc".

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/31583
The source codes are now in zephyr main tree's follow two directory in this PR:
subsys\bluetooth\host\a2dp_codec\sbc\decoder\
subsys\bluetooth\host\a2dp_codec\sbc\encoder\

## Description

I am implementing the BT A2DP function, the SBC is mandatory for A2DP.
Android uses the Bluedroid SBC encoder/decoder and maintain it.
So, I suggest to porting Android SBC encoder/decoder to zephyr, there is no need to implement one new encoder/decoder in Zephyr.
I have did one PR https://github.com/zephyrproject-rtos/zephyr/pull/31583, In this PR, the Android SBC source files are put in the follow two directories.
subsys\bluetooth\host\a2dp_codec\sbc\decoder\
subsys\bluetooth\host\a2dp_codec\sbc\encoder\
And I add one porting directory (zephyr\subsys\bluetooth\host\a2dp_codec\sbc\porting\) and put two empty files (bt_target.h and bt_types.h) to avoid changing original SBC source files and pass building.
And I did only two line codes change compared with original Android SBC codes in order to compile successfully.
The changes are: adding two "extern" to the follow two lines in dequant.c.
extern const uint32_t dequant_long_scaled[17];
extern const uint32_t dequant_long_unscaled[17];

# Dependencies

What other components does this package depend on? - No

Will the Zephyr project have a direct dependency on the component, or
will it be included via an abstraction layer with this component as a
replaceable implementation?
- I have implemented one abstraction layer in Zephyr to simplify the interfaces(APIs) and usage.

## Revision

Use the Android codes from
https://cs.android.com/android/platform/superproject/+/master:system/bt/embdrv/sbc/

## License

Apache 2.0

carlescufi, 2021-07-14:
@MarkWangChinese did you have to make changes to the original Android source code in order to make it work for Zephyr?

carlescufi, 2021-07-14:
TSC: Requires further analysis, we will invite @MarkWangChinese to the regular Bluetooth meeting.

MarkWangChinese, 2021-07-15:
> @MarkWangChinese did you have to make changes to the original Android source code in order to make it work for Zephyr?

@carlescufi I did only two line codes change compared with original Android SBC codes in order to compile successfully.
The changes are: adding two "extern" to the follow two lines in dequant.c.
extern const uint32_t dequant_long_scaled[17];
extern const uint32_t dequant_long_unscaled[17];

hongshui3000, 2021-08-28:
Any progress？

dleach02, 2021-12-06:
@MarkWangChinese has confirmed that he can serve as the maintainer of this module (per TSC action item)

MarkWangChinese, 2021-12-07:
> @MarkWangChinese has confirmed that he can serve as the maintainer of this module (per TSC action item)
> 
> Link to source updated in description

Thanks David.