zephyrproject-rtos/zephyr #36873: USB AUDIO Byte alignment issues

hongshui3000, 2021-07-12:
**Describe the bug**
I found that the usb descriptors are all single-byte aligned, but the uint16_t * pointer is used in the code to directly access the descriptor data. This will cause alignment problems, depending on the overall offset of the descriptor.

zephyr\subsys\usb\class\audio\audio.c
```
/**
 * Helper function for fixing controls in feature units descriptors.
 */
static void fix_fu_descriptors(struct usb_if_descriptor *iface)
{
	struct cs_ac_if_descriptor *header;
	struct feature_unit_descriptor *fu;

	header = (struct cs_ac_if_descriptor *)
			((uint8_t *)iface + USB_PASSIVE_IF_DESC_SIZE);

	fu = (struct feature_unit_descriptor *)((uint8_t *)header +
						header->bLength +
						INPUT_TERMINAL_DESC_SIZE);

	/* start from 1 as elem 0 is filled when descriptor is declared */
	for (int i = 1; i < get_num_of_channels(fu); i++) {
		//*(fu->bmaControls + i) = fu->bmaControls[0]; ------------------------------------>original
		((uint8_t *)(fu->bmaControls + i))[0] = ((uint8_t *)fu->bmaControls)[0]; --------------->my fix
		((uint8_t *)(fu->bmaControls + i))[1] = ((uint8_t *)fu->bmaControls)[1];
	}

	if (header->bInCollection == 2) {
		fu = (struct feature_unit_descriptor *)((uint8_t *)fu +
			fu->bLength +
			INPUT_TERMINAL_DESC_SIZE +
			OUTPUT_TERMINAL_DESC_SIZE);
		for (int i = 1; i < get_num_of_channels(fu); i++) {
			//*(fu->bmaControls + i) = fu->bmaControls[0];------------------------------------>original
			((uint8_t *)(fu->bmaControls + i))[0] = ((uint8_t *)fu->bmaControls)[0]; --------------->my fix
			((uint8_t *)(fu->bmaControls + i))[1] = ((uint8_t *)fu->bmaControls)[1];
		}
	}
}

```
```
static uint16_t get_controls(const struct feature_unit_descriptor *fu)
{
	uint16_t controls = ((uint8_t *)fu + BMA_CONTROLS_OFFSET)[0];----------------->my fix
	controls |= ((uint8_t *)fu + BMA_CONTROLS_OFFSET)[1]<<8;
	//return *(uint16_t *)((uint8_t *)fu + BMA_CONTROLS_OFFSET);---------------------->original
	return controls;
}
```

cfriedt, 2021-07-13:
@hongshui3000 - it sounds like you might have provided a fix. Are you able to make a pull request? If so, please make sure you mention this issue in the commit log - e.g. `Fixes #36873`.

hongshui3000, 2021-07-15:
Will this PR be submitted to the upstream？@jfischer-no

jfischer-no, 2021-07-15:
> Will this PR be submitted to the upstream？@jfischer-no

If you mean the patch linked above then yes, but I want to clarify if this alignment problem is to be expected with modern compilers. I do not see any difference in binary after this change.

hongshui3000, 2021-07-16:
@jfischer-no  In the xcc (arch:xtensa, gcc 4.2) I use, these codes have the issue said.

jfischer-no, 2021-07-19:
> @jfischer-no In the xcc (arch:xtensa, gcc 4.2) I use, these codes have the issue said.I haven't tried this code in other compilers.The patch you provided should solve my issue.

@hongshui3000 thanks for the info.

PR #37044 (2021-07-20): usb: audio: prevent unaligned memory access
Prevent unaligned memory access.

Fixes #36873
commit 244f69d1 (2021-07-20): usb: audio: prevent unaligned memory access

Prevent unaligned memory access.

Fixes #36873
closed 2021-07-20 by nashif