zephyrproject-rtos/zephyr #36882: MCUMGR: fs upload fail for first time file upload

ExtremeGTX, 2021-07-12:
**Describe the bug**
When a user use `mcumgr-cli` to upload a new file to the device, the upload may stuck in the middle of uploading.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b stm32f769i_disco .\samples\subsys\mgmt\mcumgr\smp_svr
2. Use the attached .config
3. west build

**Expected behavior**
Successful file upload.

**Impact**
**Workaroud:** File needs to be re-uploaded for a successful upload.

**Logs and console output**
```
[00:00:16.469,000] <err> fs: failed get file or dir stat (-2)
```

**Environment (please complete the following information):**
 - Board STM32F769i_Disco
 - Windows
 - Zephyr SDK
 - main@https://github.com/zephyrproject-rtos/zephyr/commit/882600834e80d2d3d5db4e61d72f63a8bd997fcc

**Additional context**

<details>
<summary>My .config</summary>

```
CONFIG_GPIO=y
# CONFIG_SPI is not set
# CONFIG_I2C is not set
# CONFIG_KSCAN is not set
# CONFIG_WIFI is not set
# CONFIG_MODEM is not set
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_BOARD="stm32f769i_disco"
CONFIG_SOC="stm32f769xx"
CONFIG_SOC_SERIES="stm32f7"
CONFIG_NUM_IRQS=110
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=216000000
CONFIG_HEAP_MEM_POOL_SIZE=256
CONFIG_ROM_START_OFFSET=0
# CONFIG_WATCHDOG is not set
# CONFIG_UART_NS16550 is not set
# CONFIG_PM is not set
# CONFIG_PM_DEVICE is not set
CONFIG_CORTEX_M_SYSTICK=y
CONFIG_CLOCK_CONTROL=y
CONFIG_SYS_CLOCK_TICKS_PER_SEC=10000
CONFIG_BUILD_OUTPUT_HEX=y
# CONFIG_FPU is not set
CONFIG_FLASH_SIZE=2048
CONFIG_FLASH_BASE_ADDRESS=0x8000000
# CONFIG_GPIO_MCUX is not set
# CONFIG_CPU_HAS_CUSTOM_FIXED_SOC_MPU_REGIONS is not set
# CONFIG_TINYCRYPT is not set
CONFIG_PINMUX_INIT_PRIORITY=45
CONFIG_SERIAL=y
# CONFIG_UART_PL011 is not set
CONFIG_CLOCK_CONTROL_STM32_CUBE=y
CONFIG_UART_STM32=y
CONFIG_GPIO_STM32=y
CONFIG_PINMUX_STM32=y
# CONFIG_SRAM_VECTOR_TABLE is not set

#
[... 1002 lines cut ...]
# CONFIG_BUILD_OUTPUT_S19 is not set
# CONFIG_BUILD_OUTPUT_UF2 is not set
# CONFIG_BUILD_OUTPUT_STRIPPED is not set
# CONFIG_APPLICATION_DEFINED_SYSCALL is not set
# CONFIG_MAKEFILE_EXPORTS is not set
# end of Build Options
# end of Build and Link Features

#
# Boot Options
#
# CONFIG_IS_BOOTLOADER is not set
# CONFIG_BOOTLOADER_MCUBOOT is not set
# CONFIG_BOOTLOADER_BOSSA is not set
# end of Boot Options

#
# Compatibility
#
CONFIG_COMPAT_INCLUDES=y
# end of Compatibility

```

</details>

de-nordic, 2021-07-12:
> [00:00:16.469,000] <err> fs: failed get file or dir stat (-2)

This log is result of `fs_stat` trying to figure out whether the file exists, before attempting to truncate it for the purpose of overwrite. The -2 means `-ENOENT`.

Can you run the cli with `--loglevel=debug` and see what is happening?
What error is returned by the cli?

de-nordic, 2021-07-14:
@ExtremeGTX Any new info on the issue?

ExtremeGTX, 2021-07-14:
Sorry for the late reply, I didn't got the notification.
[mcumgr_fs_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/6814296/mcumgr_fs_log.txt)

I think it is because of logger writing the `fs_stat` error to the shell.

de-nordic, 2021-07-14:
Are you using the same UART for mcumgr and for console?

ExtremeGTX, 2021-07-14:
yes, Console, shell and MCUMGR

de-nordic, 2021-07-14:
> 
> 
> yes, Console, shell and MCUMGR

OK, so if the console is log sink and at the same time used for mcumgr, then you may get your transfer interrupted by other messages, with the same result.

ExtremeGTX, 2021-07-14:
so, there is no options for making Console and MCUMGR live together ? like disable log to the console if a file transfer is in progress

de-nordic, 2021-07-14:
You could probably use this https://docs.zephyrproject.org/latest/reference/logging/index.html?highlight=log#c.log_backend_enable, but honestly I have not tried that.
@nordic-krch  what do you think about this?

nordic-krch, 2021-07-14:
yes, you can temporarily disable logging to given backend with log_backend_disable. For shell you can use `z_shell_log_backend_disable()` which wraps around `log_backend_disable` but also changes internal state,

ExtremeGTX, 2021-07-14:
so you suggest to use `log_backend_disable` when MCUMGR receives a `fs` command then call `log_backend_enable` after finish.
Would you welcome a PR with this change ? it will also include a `CONFIG_` option to enable/disable this behavior .

de-nordic, 2021-07-14:
> 
> 
> so you suggest to use `log_backend_disable` when MCUMGR receives a `fs` command then call `log_backend_enable` after finish.
> Would you welcome a PR with this change ? it will also include a `CONFIG_` option to enable/disable this behavior .

The PRs are always welcome, the review may take time, so do not be discouraged.  The `CONFIG_` enabling the option is desired.

ExtremeGTX, 2021-07-26:
I tried to add `log_backend_enable` and `log_backend_disable` in the following function `smp_shell_tx_pkt` https://github.com/zephyrproject-rtos/zephyr/blob/532a85ccd978290aa5ea08bc32d74ff200d27bdd/subsys/mgmt/mcumgr/smp_shell.c#L178

but the problem is the overhead caused by `backend_filter_set` because it loops over all logging sources and same for enable.

What could be the alternative ? is there a way to get a notification from MCUMGR that transmission is started/done ? where i can enable/disable shell log backend instead of doing this per packet ?

nordic-krch, 2021-07-27:
you can try `log_backend_deactivate` which just halts the backend.

ExtremeGTX, 2021-07-27:
So I used the following code snippet
```
static int smp_shell_tx_pkt(struct zephyr_smp_transport *zst,
			    struct net_buf *nb)
{
	int rc;
	const struct shell *const sh = shell_backend_uart_get_ptr();

	log_backend_deactivate(sh->log_backend->backend);
	rc = mcumgr_serial_tx_pkt(nb->data, nb->len, smp_shell_tx_raw, NULL);
	mcumgr_buf_free(nb);

	log_backend_activate(sh->log_backend->backend, sh->log_backend->backend->cb->ctx);
	return rc;
}
```

The problem is the UART prompt, for some reason it got printed to the UART after `log_backend_activate`, if i removed the call to `log_backend_activate` things will work as expected!.

But the problem is I can't see a way of identifying the end of MCUMGR transmission in this function `smp_shell_tx_pkt`

```
time="2021-07-27 10:56:02.432" level=debug msg="Tx serial
00000000  0a                                                |.|
"
time="2021-07-27 10:56:02.46" level=debug msg="Rx serial:
00000000  1b 5b 38 44 1b 5b 4a 5b  30 30 3a 30 30 3a 33 39  |.[8D.[J[00:00:39|
00000010  2e 36 36 33 2c 30 30 30  5d 20 1b 5b 31 3b 33 31  |.663,000] .[1;31|
00000020  6d 3c 65 72 72 3e 20 66  73 3a 20 66 61 69 6c 65  |m<err> fs: faile|
00000030  64 20 67 65 74 20 66 69  6c 65 20 6f 72 20 64 69  |d get file or di|
00000040  72 20 73 74 61 74 20 28  2d 32 29 1b 5b 30 6d     |r stat (-2).[0m|
"
time="2021-07-27 10:56:02.468" level=debug msg="Tx serial
00000000  04 14                                             |..|
"
time="2021-07-27 10:56:02.468" level=debug msg="Tx serial
00000000  51 76 62 45                                       |QvbE|
"
time="2021-07-27 10:56:02.468" level=debug msg="Tx serial
00000000  0a                                                |.|
"
time="2021-07-27 10:56:02.502" level=debug msg="Rx serial:
00000000  1b 5b 31 3b 33 32 6d 75  61 72 74 3a 7e 24 20 1b  |.[1;32muart:~$ .|
00000010  5b 6d 1b 5b 38 44 1b 5b  4a 1b 5b 31 3b 33 32 6d  |[m.[8D.[J.[1;32m|
00000020  75 61 72 74 3a 7e 24 20  1b 5b 6d 06 09 41 42 55  |uart:~$ .[m..ABU|
00000030  44 41 41 41 4e 41 41 68  54 41 4c 39 69 63 6d 4d  |DAAANAAhTAL9icmM|
00000040  41 59 32 39 6d 5a 68 6b  48 67 76 39 6d 36 41 3d  |AY29mZhkHgv9m6A=|
00000050  3d                                                |=|
"
```

de-nordic, 2021-07-27:
> 
> But the problem is I can't see a way of identifying the end of MCUMGR transmission in this function `smp_shell_tx_pkt`

That might be tougher to catch. The shell does not know what it is passing from mcumgr and does not really care, and due to how the entire thing is organized the mcumgr does not bother to notify shell that this is final packet of transaction.

ExtremeGTX, 2021-07-29:
@de-nordic I opened a draft PR, so we can have a better discussion about the fix.
Thanks for your support.

PR #50945 (2022-10-11): mgmt: mcumgr: Add fs_mgmt log output workaround on file write
    If a file write was attempted with a file that does not exist, over
    the shell/UART when logging was enabled, it would output a fs_unlink
    error, this works around the issue by checking if the file exists and
    needs truncating before performing that action. It also imrproves
    flash endurance slightly by performing a truncate operation instead
    of a delete, but will fall back to a delete if the truncation
    operation fails. This issue can be also be mitigated by altering
    logging settings or adjusting the SMP  [... cut]