zephyrproject-rtos/zephyr #36885: Update ISO API to better support TWS

Thalley, 2021-07-12:
**Is your enhancement proposal related to a problem? Please describe.**
The current API lets the user bind one or more ACL connections to one or more (unconnected) CIS (iso channels). This works fine as long as the TWS setup process is

1. ACL connect device 1
2. ACL connect device 2
3. ISO bind the ACL to ISO channels (creating the CIG)
4. ISO connect both devices

But will not support 

1. ACL connect device 1
2. ISO connect device 1
3. ACL connect device 2
4. ISO connect device 2 in the same CIG

and consequently won't be able to support the scenario where one of the remove device disconnects the ACL or CIS, and we want to reconnect, as the current API is "connect all once" and does not support connecting a subset of ISO channels in a CIG. 

**Describe the solution you'd like**
To support TWS, or in general situations where are may only be partially ACL connected, we need to decouple ACL from the CIG. Such a decoupling may also help to make the broadcast (non-connected) and unicast (connected) ISO work better together with the same API. 

**Describe alternatives you've considered**
An alternative solution would be to update the `bt_iso_chan_bind` to support just binding ACL and ISO channels to existing channels after reconnection, thus not creating a new CIG. This feels more like a short-term patchy solution, rather than a proper solution. 

**Additional context**
The ISO API concept of linking channels are implicitly creating ISO channel groups, but the implementation seems only halfway complete, and the number of links is that of a complete graph becomes `(n(n-1))/2` links, where `n` is the number of channels.

Thalley, 2021-07-12:
My suggestion to update the API would be to remove the concept of linking channels, and replace it by a `bt_iso_create_group` API, which would be responsible for creating the CIG/BIG, keep that completely separate from ACL connections. Some of the other API functions would then work on groups, rather than a single channel (similar to how some of the broadcast API functions are implemented). 

The `bt_iso_chan_bind` would thus also be removed, and `bt_iso_chan_connect` would be changed to connect an ISO channel with an ACL. 

Besides being able to solve the issue with TWS (and disconnects), this will also more closely follow the HCI spec, possibly making the implementation easier to follow and/or simpler. 

This change will of course also affect how the upcoming LE Audio stream API will work, but in my opinion this would make that API better as well. The LE Audio API will also need to support the concept of subgroups, so introducing a `group` API would possibly make that easier.

Casper-Bonde-Bose, 2021-07-19:
@Thalley I generally like the idea of keeping the API close to how the HCI layer works, as you then have some very good documentation of how it works, which makes it easier for new-comers to maintain the code. The concept of creating a group makes very good sense to me, as this is also the terminology used in the specifications. 
I fully support this change request and will be happy to provide feedback on an API header-file describing the new API before the implementation is done.

Thalley, 2021-07-21:
Also would like to get some input, or at least be able to notify @alexsven and @koffes as any API update may impact your work.

rick1082, 2021-07-22:
Please allow me to think out loudly in here.
For TWS solution which involving two ACLs and two ISO CIS in one CIG, there could be following user scenario:

TWS setup (like Emil mentioned in the begin of the thread)
1. Central connects to peripheral A through ACL
2. Central connects to peripheral B through ACL
3. Central creating a CIG which contains ISO channels to peripheral A and peripheral B

If one ISO channel dropped, needs to recover the dropped ISO channel
1. Peripheral A dropped ISO channel, but ACL connection still remaining
2. Central needs to maintain the ISO channel for peripheral B
3. Central connects the ISO channel to peripheral A, which under the same CIG with peripheral B

If one ACL channel dropped, needs to recover the dropped ACL channel, also the ISO channel
1. Peripheral A dropped ACL channel, so ISO channal also dropped
2. Central needs to maintain the ISO channel for peripheral B
3. Central scans and connects with peripheral A by ACL
4. Central connectes the ISO channel to peripheral A, which under the same CIG with peripheral B

If there's no audio streaming, for low down the power consumption
1. Central terminates the CIG for peripheral A and peripheral B
2. Central still maintaining the ACL connection between peripheral A and peripheral B
3. Central start the CIG wth peripheral A and B again for the incoming audio stream

I also agree that we can use APIs which close to HCI commands, which is good to track the procedure with HCI commands for specification. Involving "Group" concept might also help us to maintain a CIG. If an ISO channel dropped, we can add it back to the existed CIG.

Casper-Bonde-Bose, 2021-07-22:
@rick1082 I was actually of the impression that one could not detects loss of a CIS connection, but only the ACL by use of the link supervision timeout. Have I missed something here?
I think the intension in BLE Audio is that the audio source shall keep transmitting the CIS data as long as the ACL is connected - this allows audio data to flow even in the cases where the ACK packets are lost - which can happen when there is a significant difference in TX power of the audio source and audio sink devices - the typical case for a phone and a TWIE device. It would allow the ACL to use for example 1mbit or coded phy while the CIS uses 1 or 2MBit.
But I do agree in the ACL dropped case, but also the connect to device 1 first, and then when device 2 is found you attach that as well. This will allow for 1) the user might only have power on one of the TWIE devices even though there are two in the set. 2) the connection/stream to device 1 can be setup immediately when it is found, potentially speeding up the total connection time.

rick1082, 2021-07-22:
Please let me add the case number in the previous comment :P
@Casper-Bonde-Bose, for the case(2), there could be a chance that the central failed to create a CIG contains two CIS for peripheral A and peripheral B, but only create a CIG with one CIS. So we might need to detect and recover it.

Thalley, 2021-07-23:
> @rick1082 I was actually of the impression that one could not detects loss of a CIS connection, but only the ACL by use of the link supervision timeout. Have I missed something here?

Each iso channel (CIS or BIS) has a unique connection handle in the same space as ACL connection handles, and each CIS (and in theory, but not in practice also BIS) can be disconnected individually. When a CIS is connected, we get the same Disconnect Complete event as we would for an ACL. 

In the case that we have 2 CIS to a single device, one of these CIS may also disconnect without the other CIS or the ACL, and the current API also does not support reconnecting this from the user. 

> I think the intension in BLE Audio is that the audio source shall keep transmitting the CIS data as long as the ACL is connected

Remember that ISO != Audio. For LE Audio, the behaviour is defined in upper layers.

Thalley, 2021-07-23:
Some time after next I'll start to design an API for ISO that should solve some of the challenges mention in this.

Thalley, 2021-08-09:
Given the similarities with create_cig/create_big, we could have a single `create_iso_group` with a type parameter, and a single `terminate_iso_group` as well if we wanted to streamline them even more.

PR #37778 (2021-08-24): Bluetooth: ISO: Update connected ISO API
Update the connected ISO API to be more
similar to the broadcast ISO API as well
as the HCI spec.

This updated API allows for more flexibility
and will better support scenarios such as true
wireless setup, as ISO channels and connections
are more independent now.

fixes #36885
closed 2021-08-24 by cfriedt