zephyrproject-rtos/zephyr #36889: string.h / strcasestr()

cederom, 2021-07-12:
Hello world :-)

It would be really nice to have `strcasestr()` function in standard Zephyr C library and `string.h`:-)

Maybe Zephyr could use such functions from the build host OS when not available in Zephyr? I know that may introduce compatibility problems but for testing before they are available in the upstream that would be some sort of solution :-)

Thank you :-)
Tomek

cfriedt, 2021-07-12:
Hi @cederom 

The reentrant `strtok_r(3)` is available in Zephyr's minimal C library [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/libc/minimal/source/string/string.c#L180) and it is highly encouraged to use that instead of `strtok(3)` to eliminate race conditions.

Similarly, `strncasecmp(3)` is available in Zephyr's minimal C library [here](https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/lib/libc/minimal/source/string/strncasecmp.c#L11). Using the `n` variants of common string functions is preferred for most applications.

If those are insufficient for your needs, please consider opening a PR.

closed 2021-07-12 by cfriedt

cederom, 2021-07-13:
Thank you @cfriedt this is exactly what I need :-) 

If this is internal Zephyr implementation why does not start it with `k_` or `z_` prefix? Like `k_msleep()`? :-)