zephyrproject-rtos/zephyr #36891: Significant TCP perfomance regression

xhpohanka, 2021-07-12:
**Describe the bug**
A commit #c6fac25e87323e86540674166af40c0a4da767be introduces significant drop in TCP performance for my application which is a web server built around civetweb.

With this commit data buffer allocations fails often. RX direction is worse than TX. Eg. file upload is more than 4times slower.

```
[00:00:03.571,000] <err> net_pkt: Data buffer (576) allocation failed.
[00:00:03.672,000] <err> net_pkt: Data buffer (68) allocation failed.
```

Tomorrow (hopefully) I will try to share more details.

cfriedt, 2021-07-13:
@rlubos - would you be able to verify this regression?
@xhpohanka - can you also specify the value you have for `CONFIG_NET_LOG_LEVEL`?

xhpohanka, 2021-07-13:
This issue is independent to LOG_LEVEL, but i have there INF usually.

I also forgot to mention that I see this on stm32h7 platform.

rlubos, 2021-07-13:
We had reports of the [decreased UDP performance](https://github.com/zephyrproject-rtos/zephyr/issues/36606), but it was rather related to the extra mutex protection added to the net stack, not the buffer allocation.

Actually, the only change that the commit you mention introduced, is the buffer pool used for allocation. `net_pkt_alloc_with_buffer()` is just a thin wrapper over `pkt_alloc_with_buffer()`, which enforces the use of the TX pool. With the changes from the aforementioned commit, the clone operation will use the pool from which the packet originated, so after this change the RX pool could be more heavily occupied. Perhaps it's just a matter of increasing the number of RX buffers available to restore the original behaviour? At the same time, it should theoretically be possible to decrease the number of TX buffers, as there should be less allocations from the TX pool.

xhpohanka, 2021-07-13:
> Perhaps it's just a matter of increasing the number of RX buffers available to restore the original behaviour? At the same time, it should theoretically be possible to decrease the number of TX buffers, as there should be less allocations from the TX pool.

I will check it, thanks

xhpohanka, 2021-07-13:
It seems a bit magic to me now but with `net_pkt_alloc_with_buffer()` everything is fine but with `pkt_alloc_with_buffer(pkt->slab,..)` the allocations are unsuccessfull with no respect to `NET_BUF_RX_COUNT` and `NET_BUF_TX_COUNT` size. I do not have unlimited SRAM, but I tried to double both with no progress...

```
00:00:35.055,000] <err> net_pkt: Data buffer (590) allocation failed.
[00:00:35.055,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:00:35.062,000] <dbg> testapp.check_header: HeaderCrc: Header:0x9494794d  Data:0x9494794d
[00:00:35.062,000] <dbg> testapp.field_get_fw: size: 453344
[00:00:35.079,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 4096, remaining = 449312
[00:00:35.103,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 8192, remaining = 445216
[00:00:35.121,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 12288, remaining = 441120
[00:00:35.205,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 16384, remaining = 437024
[00:00:35.212,000] <err> net_pkt: Data buffer (590) allocation failed.
[00:00:35.212,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:00:35.228,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 20480, remaining = 432928
[00:00:35.247,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 24576, remaining = 428832
[00:00:35.266,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 28672, remaining = 424736
[00:00:35.286,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 32768, remaining = 420640
[00:00:35.351,000] <err> net_pkt: Data buffer (590) allocation failed.
[00:00:35.351,000] <err> eth_stm32_hal: Failed to obtain RX buffer
...
```

rlubos, 2021-07-15:
So perhaps there's a memory leak somewhere? There is also a net shell utility that helps to track allocations, perhaps it could thell where all the buffers gone (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/net_shell.c#L5998).

xhpohanka, 2021-07-15:
It seems like there isn't to much things allocated :D

```
uart:~$ net mem
Fragment length 128 bytes
Network buffer pools:
Address         Total   Avail   Name
0x24012554      256     253     RX
0x2401258c      256     253     TX
0x2401282c      96      96      RX DATA (rx_bufs)
0x24012860      192     192     TX DATA (tx_bufs)
[00:01:04.746,000] <err> net_pkt: Data buffer (590) allocation failed (eth_rx:391)
[00:01:04.747,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:01:04.848,000] <err> net_pkt: Data buffer (576) allocation failed (net_pkt_clone:1772)
[00:01:05.066,000] <err> net_pkt: Data buffer (576) allocation failed (net_pkt_clone:1772)
[00:01:03.946,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:01:04.746,000] <err> net_pkt: Data buffer (590) allocation failed (eth_rx:391)
[00:01:04.747,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:01:04.848,000] <err> net_pkt: Data buffer (576) allocation failed (net_pkt_clone:1772)
[00:01:05.066,000] <err> net_pkt: Data buffer (576) allocation failed (net_pkt_clone:1772)
[00:01:05.259,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 32768, remaining = 397904
[00:01:05.286,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 36864, remaining = 393808
[00:01:05.324,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 40960, remaining = 389712
uart:~$ net allocs
Network memory allocations

memory          Status  Pool    Function alloc -> freed
0x24061ba8/1     used      RX   tcp_conn_alloc():1221
0x24065ba8/1     used      TX   tcp_conn_alloc():1229
0x24061b68/1     used      RX   tcp_conn_alloc():1221
0x24065b68/1     used      TX   tcp_conn_alloc():1229
0x24061ae8/1     used      RX   tcp_conn_alloc():1221
0x240658a8/1     used      TX   tcp_conn_alloc():1229
0x240615e8       free      RX   eth_rx():391 -> processing_data():159
0x240618a8       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x24065a68       free      TX   tcp_out_ext():811 -> tcp_send():340
0x240665d8       free   TDATA   ethernet_fill_header():486 -> ethernet_remove_l2_header():566
0x24061a68       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x240619e8       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x240617e8       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x240616e8       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x24061b28       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x24061768       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x24061628       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x24061668       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x240616a8       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x24061a28       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x24061aa8       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x24061868       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x240617a8       free      RX   eth_rx():391 -> processing_data():159
0x240619a8       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x240615e8       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x24061968       free      RX   net_pkt_clone():1772 -> zsock_recv_stream():1201
0x240665d8       free   TDATA   ethernet_fill_header():486 -> ethernet_remove_l2_header():566
0x24065968       free      TX   tcp_out_ext():811 -> tcp_send():340
0x24066710       free   TDATA   ethernet_fill_header():486 -> ethernet_remove_l2_header():566
0x240617a8       free      RX   eth_rx():391 -> processing_data():159
0x24065aa8       free      TX   tcp_out_ext():811 -> tcp_send():340
0x24065aa8       free      TX   tcp_out_ext():811 -> tcp_send():340
[00:01:06.384,000] <err> net_pkt: Data buffer (590) allocation failed (eth_rx:391)
[00:01:06.385,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:01:07.139,000] <err> net_pkt: Data buffer (590) allocation failed (eth_rx:391)
[00:01:07.139,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:01:07.165,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 49152, remaining = 381520
[00:01:07.178,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 53248, remaining = 377424
[00:01:07.190,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 57344, remaining = 373328
[00:01:07.203,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 61440, remaining = 369232
[00:01:05.354,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 45056, remaining = 385616
[00:01:06.384,000] <err> net_pkt: Data buffer (590) allocation failed (eth_rx:391)
[00:01:06.385,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[... 114 lines cut ...]
[00:01:11.334,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 126976, remaining = 303696
[00:01:12.057,000] <err> net_pkt: Data buffer (590) allocation failed (eth_rx:391)
[00:01:12.057,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:01:12.326,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 131072, remaining = 299600
[00:01:12.339,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 135168, remaining = 295504
[00:01:12.353,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 139264, remaining = 291408
[... cut ...]
[00:01:13.655,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:01:13.693,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 147456, remaining = 283216
[00:01:13.706,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 151552, remaining = 279120
[00:01:13.718,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 155648, remaining = 275024
[00:01:13.730,000] <dbg> testapp.field_get_fw: firmware sector written (0), bytes = 159744, remaining = 270928
uart:~$ 
```

rlubos, 2021-07-15:
So if there are so many buffers available, why does it fail to allocate them? And why does it only happen with changes from c6fac25 ? Some bug in the net_buf allocation algorithm? Hard to tell actually w/o debugger attached to the actual hardware and application.

xhpohanka, 2021-07-15:
I will try to prepare some minimal example.

xhpohanka, 2021-07-16:
Minimal example is here: https://github.com/xhpohanka/civetweb_upload

It is just slightly altered zephyr's civetweb sample with additional handler for file upload. I'm testing it with similar command `curl --form file='@fw.bin' http://192.0.2.1/fwupdate`

It behaves exactly as described in previous comments. I was not able to see it in qemu. Could it be possible that there is some concurrency issue? Maybe RX packets are not freed as quick as TX ones...

rlubos, 2021-07-16:
> I was not able to see it in qemu

That could suggest that there's some issue with the particular Ethernet driver? I'm not familiar with `eth_stm32_hal.c` driver, but what I don't like is that it tries to allocate the buffer, w/o a timeout:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ethernet/eth_stm32_hal.c#L392

The question is what are the priorities of the Ethernet driver RX thread, and the application thread that consumes the data. If the driver thread has a higher priority, and the data arrives faster than the application thread can consume it (which I guess is waht we observe based on the above log), the Ethernet thread will eventually start to drop the data, unless it gives the application thread a chance to free some buffers (by specifying the allocation timeout). Otherwise, frame is dropped silently, and we get significant perfomance drop, as we get gaps in the TCP stream.

xhpohanka, 2021-07-20:
>  but what I don't like is that it tries to allocate the buffer w/o a timeout

That is an interresting notice. Changing the `net_pkt_rx_alloc_with_buffer()` allocation within the driver to wait fixes my issue. The wait period has to be quite large (100ms) to prevent the unsuccessfull allocations completely. I checked the rest of drivers and they use `K_NO_WAIT` too, but all of them do the allocation inside an interrupt. STM32 is the only one with separate RX thread.

> The question is what are the priorities of the Ethernet driver RX thread, and the application thread that consumes the data

ethernet driver default priority is `-14` while the application (server) is in preempt range (0)

rlubos, 2021-07-20:
I suggest then fixing the issue on the driver side to improve the driver reliability.

rlubos, 2021-08-04:
@xhpohanka Any update on this, will you be able to submit a patch to the driver that fixed the performance issue for you?

PR #37459 (2021-08-05): drivers: ethernet: eth_stm32_hal.c: add wait time for buffer allocation
Driver's RX thread can wait some time before a free buffer is available.
When resources are limited we can easily run out of them.

Fixes #36891.
commit 1df05690 (2021-08-05): drivers: ethernet: eth_stm32_hal.c: add wait time for buffer allocation

Driver's RX thread can wait some time before a free buffer is available.
When resources are limited we can easily run out of them.

Fixes #36891.
closed 2021-08-05 by cfriedt