zephyrproject-rtos/zephyr #36905: Improve (k_)malloc and heap documentation

carlescufi, 2021-07-12:
In Zephyr there are two kinds of heaps:

- kernel-provided heaps
- libc-provided heap

Kernel-provided heaps (among them system heap) are documented here: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/reference/kernel/memory/heap.html#heap-v2

As for libc-provided heaps, matters are more complicated. First of all there are several types of C library (among them minimal libc and newlib). These libraries have their own heap that provides memory for malloc() (the kernel-provided system heap provides memory for k_malloc()).

How this libc-provided heap is configured depends on many variables: MPU, userspace, MMU etc.

What would be necessary is to have a `(k_)malloc` guide or section, that describes where and how memory is allocated depending on which options and APIs are used.

stephanosio, 2021-07-13:
TODO (RFC):

1. Add memory management-related section under [API Reference-C standard library](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/reference/libc/index.html) covering both minimal C library and newlib.
2. Add "Dynamic Memory Management" (or just "Memory Management") section under [User and Developer Guides](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/guides/index.html) describing different dynamic memory management mechanisms available (kernel and libc).

ycsin, 2022-02-07:
Any update to this? Is there any rule of thumb as in when to use `malloc` / `k_malloc`?

PR #46077 (2022-06-01): Toolchains and language support documentation rework
This series reworks the toolchain and language support-related documentations with various documentation enhancements and bug fixes.

Closes #36905
Fixes #43636 
Fixes #32885

Docs preview: https://builds.zephyrproject.io/zephyr/pr/46077/docs/
closed 2022-06-01 by carlescufi