zephyrproject-rtos/zephyr #36909: Shell `log` commands crash the system if `CONFIG_SHELL_LOG_BACKEND` is not defined

brycewilkins, 2021-07-12:
**Describe the bug:**
When accessing the shell over RTT channel 0 and the log backend over RTT channel 1, some `log` shell commands (e.g. `log status`, `log go`, etc.) will crash the system if the shell log backend (`CONFIG_SHELL_LOG_BACKEND`) is not configured.

It is easy to avoid the problem by not executing some `log` commands, however a new user of Zephyr RTOS is unlikely to be aware of such things and will likely have a bad initial experience.

**Diagnosis:**
I found that in `log_cmds.c` the function `shell_state_precheck(...)` does not check `log_backend` pointer before dereferencing:
```c
static bool shell_state_precheck(const struct shell *shell)
{
	if (shell->log_backend->control_block->state
				== SHELL_LOG_BACKEND_UNINIT) {
...
```
and in `shell_log_backend.c` the pointer is defined as `NULL`.
https://github.com/zephyrproject-rtos/zephyr/blob/9bc9e15852559fd26f48882c92c4642d1ef0456f/include/shell/shell_log_backend.h#L103

Debugger shows that `log_backend` does not point to a valid `control_block`:

[image]

**To Reproduce:**
1. Configure a project to use RTT for the shell and log backends with following CONFIGs:
```
CONFIG_USE_SEGGER_RTT=y

CONFIG_SHELL=y
CONFIG_SHELL_BACKEND_RTT=y
CONFIG_SHELL_LOG_BACKEND=n

CONFIG_LOG=y
CONFIG_LOG_BACKEND_RTT=y
CONFIG_LOG_BACKEND_RTT_BUFFER=2
```
2. Ensure `CONFIG_SHELL_LOG_BACKEND` is not defined (as shown above).
2. Compile and flash a target.
4. After connecting to the shell run one of the `log` commands, e.g. `log status`.
5. The system will crash.

**Expected behavior:**
No output and the shell prompt should return, or an error message (that the shell log backend is not enabled).

**Impact:**
The bug crashes the system and this is overall a bad user experience.

**Logs and console output:**
Example of working command:
```
rtt$ log backend
backend - Logger backends commands.
Subcommands:
  log_backend_rtt  :
rtt$
```

Example of command that crashes the system:
```
rtt$ log go
<no output; hard fault>
```

**Environment:**
 - OS: MacOS
 - Toolchain: GCC 10.2.1
 - Commit: v2.6.0

cfriedt, 2021-07-13:
@brycewilkins - it sounds as though you have identified the root cause. Would you be willing to make a PR to fix it?

brycewilkins, 2021-07-13:
@cfriedt Yes, I can put up a PR to fix it.

nordic-krch, 2021-07-13:
@brycewilkins i've disabled those commands when logging is not enabled.

closed 2021-07-13 by cfriedt

brycewilkins, 2021-07-13:
> @brycewilkins i've disabled those commands when logging is not enabled.

Yeah, I discovered yesterday that `CONFIG_LOG_CMDS=n` masks the problem, but it has to be done explicitly (`CONFIG_LOG_CMDS` will default to `y` if SHELL is enabled).

Thanks @nordic-krch for the fix!

PR #36912 (2021-07-13): logging: Shell commands available when shell acts as log backend
Some commands shell not be present when shell is not acting as
a log backend. Use of them lead to crash in that case.

Two commands remains `list_backends` and `backend` which allow to control other log backends from shell.

Fixes #36909.