zephyrproject-rtos/zephyr #36917: Runtime device PM is broken on STM32

gmarull, 2021-07-13:
**Describe the bug**

@erwango reported that device PM related tests started to fail on STM32 targets after the introduction of device PM enum states cc2f0e9c08dcad59a4cc0117028a5fcf9d73f433. After some investigation, it turns out that the change to enums uncovered an existing issue on STM32 pinmux driver.

Prior to the enum changes, we had:
https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.6.0/include/pm/device.h#L27-L33

and we now have:
https://github.com/zephyrproject-rtos/zephyr/blob/main/include/pm/device.h#L27-L33

That is, the `PM_DEVICE_STATE_ACTIVE` was `1` and now is `0`. Making the enum start at `1` fixed the problem, something that sounded like an uninitialized variable.

The problem is the following:
1. UART(s) are initialized before GPIO drivers
2. UART makes use of the STM32 pinmux API to set up pinmux
3. The STM32 pinmux driver, in turn, makes use of the GPIO driver (PM only) without checking for it to be ready
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/pinmux/pinmux_stm32.c#L129-L133
4. The GPIO driver pm_control function gets invoked, however, the initial pm_state is now `0` (static driver data is initialized to zeroes) which matches with the current value for `PM_DEVICE_STATE_ACTIVE`. Because the driver PM function only operates if there is a state change, the GPIO block is never enabled and so the serial port does not output anything, making the test to timeout. Note: If the test is debugged, one can verify that it passes, but nothing is _printed_.

I'm not sure what is the best solution here, but:
1. Any device needs to be checked for readiness before operation. Making use of it without this check, even for PM, leads to undefined behavior.
2. Drivers should not care about the runtime PM status of other devices. In this case, either the pinmux driver needs to be PM-aware or it needs to use a GPIO API that will, in turn, enable/disable GPIO.

**To Reproduce**

Any test with `CONFIG_PM_DEVICE_RUNTIME=y` will fail, e.g.

```
./scripts/twister -T tests/subsys/pm/power_mgmt --device-testing --device-serial /dev/ttyACM0 -p disco_l475_iot1
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**

Runtime device PM is broken on STM32.

**Logs and console output**

N/A

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: `main` or cc2f0e9c08dcad59a4cc0117028a5fcf9d73f433

**Additional context**
N/A

erwango, 2021-07-13:
One thing we'll need to do to fix this is to move gpio init before any gpio users such as uart (through pinmux):
So, in gpio devices init macro:https://github.com/zephyrproject-rtos/zephyr/blob/a927ae39eda5ebc125437f3cee8fd741f0d161d5/drivers/gpio/gpio_stm32.c#L685-L690
`POST_KERNEL` should be replaced by `PRE_KERNEL_1`.

This change functionally fixes the issue.
Though, as mentioned, from architecture point of view, we also need to implement a better way to call gpio functions from pinmux pseudo driver.

erwango, 2021-07-15:
@gmarull question a bout this part:

> The GPIO driver pm_control function gets invoked, however, the initial pm_state is now 0 (static driver data is initialized to zeroes) which matches with the current value for PM_DEVICE_STATE_ACTIVE

Is that actually correct to initialize pm_state to `PM_DEVICE_STATE_ACTIVE` in PM_DEVICE_RUNTIME?
I would expect they are initialized to `PM_DEVICE_STATE_OFF` and moved to `PM_DEVICE_STATE_ACTIVE` only in case there is a user.

PR #36988 (2021-07-16): STM32: Fix runtime device PM
Runtime device PM was broken on STM32 following merge of https://github.com/zephyrproject-rtos/zephyr/commit/cc2f0e9c08dcad59a4cc0117028a5fcf9d73f433 that unveiled a flaw in current
STM32 PM implementation around gpio/pinmux.

Fix this issue by changing initialization step of gpio devices and reworking calls to gpio in pinmux
pseudo driver.

Fixes #36917
closed 2021-07-16 by cfriedt