zephyrproject-rtos/zephyr #36926: samples/boards/nrf/system_off wouldn't compile on Particle Xenon board

koutaiba95, 2021-07-13:
**Describe the bug**
The sample for power measurement on nrf platform wouldn't compile on Particle Xenon. The compiler doesn't recognize button0 in lines 60 and 62 of the main.
As i have little experience i couldn't resolve the issue without consulting with someone. He pointed out to the mesh_feather.dtsi file in the dts folder of the board prticle_xenon. 
Replacing button0 with its alias : mode_button defined in line 54 of the mesh_feather.dtsi resolved the issue.

**To Reproduce**
Steps to reproduce the behavior:
I created a bin folder next to the source folder inside the system_off folder
the i use the following command from a terminal: west build -b particle_xenon -s system_off/  -d system_off/bin

**Expected behavior**
sucessful compiling

**Impact**
As a beginner in using zephyr and embedded OS i was trying to flash this sample to try it and try to understand it. I couldn't find an answer for the issue and without the help of someone more experienced than me i wouldn't have found the cause.

**Logs and console output**
`../src/main.c: In function 'main':
/home/koutaiba/zephyrproject/zephyr/include/devicetree.h:173:36: error: 'DT_N_NODELABEL_button0_P_gpios_IDX_0_VAL_pin' undeclared (first use in this function); did you mean 'DT_N_S_gpio_keys_S_button_0_P_gpios_IDX_0_VAL_pin'?
  173 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                                    ^~~~~~~~~~~~~~~
/home/koutaiba/zephyrproject/zephyr/include/devicetree.h:2326:24: note: in definition of macro 'DT_CAT'
 2326 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
/home/koutaiba/zephyrproject/zephyr/include/devicetree.h:962:2: note: in expansion of macro 'DT_PROP'
  962 |  DT_PROP(node_id, pha##_IDX_##idx##_VAL_##cell)
      |  ^~~~~~~
/home/koutaiba/zephyrproject/zephyr/include/devicetree/gpio.h:158:2: note: in expansion of macro 'DT_PHA_BY_IDX'
  158 |  DT_PHA_BY_IDX(node_id, gpio_pha, idx, pin)
      |  ^~~~~~~~~~~~~
/home/koutaiba/zephyrproject/zephyr/include/devicetree/gpio.h:169:2: note: in expansion of macro 'DT_GPIO_PIN_BY_IDX'
  169 |  DT_GPIO_PIN_BY_IDX(node_id, gpio_pha, 0)
      |  ^~~~~~~~~~~~~~~~~~
../src/main.c:60:21: note: in expansion of macro 'DT_GPIO_PIN'
   60 |  nrf_gpio_cfg_input(DT_GPIO_PIN(DT_NODELABEL(button0), gpios),
      |                     ^~~~~~~~~~~
/home/koutaiba/zephyrproject/zephyr/include/devicetree.h:173:29: note: in expansion of macro 'DT_CAT'
  173 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                             ^~~~~~
../src/main.c:60:33: note: in expansion of macro 'DT_NODELABEL'
   60 |  nrf_gpio_cfg_input(DT_GPIO_PIN(DT_NODELABEL(button0), gpios),
      |                                 ^~~~~~~~~~~~
/home/koutaiba/zephyrproject/zephyr/include/devicetree.h:173:36: note: each undeclared identifier is reported only once for each function it appears in
  173 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                                    ^~~~~~~~~~~~~~~
/home/koutaiba/zephyrproject/zephyr/include/devicetree.h:2326:24: note: in definition of macro 'DT_CAT'
 2326 | #define DT_CAT(a1, a2) a1 ## a2
      |                        ^~
/home/koutaiba/zephyrproject/zephyr/include/devicetree.h:962:2: note: in expansion of macro 'DT_PROP'
  962 |  DT_PROP(node_id, pha##_IDX_##idx##_VAL_##cell)
      |  ^~~~~~~
/home/koutaiba/zephyrproject/zephyr/include/devicetree/gpio.h:158:2: note: in expansion of macro 'DT_PHA_BY_IDX'
  158 |  DT_PHA_BY_IDX(node_id, gpio_pha, idx, pin)
      |  ^~~~~~~~~~~~~
/home/koutaiba/zephyrproject/zephyr/include/devicetree/gpio.h:169:2: note: in expansion of macro 'DT_GPIO_PIN_BY_IDX'
  169 |  DT_GPIO_PIN_BY_IDX(node_id, gpio_pha, 0)
      |  ^~~~~~~~~~~~~~~~~~
../src/main.c:60:21: note: in expansion of macro 'DT_GPIO_PIN'
   60 |  nrf_gpio_cfg_input(DT_GPIO_PIN(DT_NODELABEL(button0), gpios),
      |                     ^~~~~~~~~~~
/home/koutaiba/zephyrproject/zephyr/include/devicetree.h:173:29: note: in expansion of macro 'DT_CAT'
  173 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                             ^~~~~~
../src/main.c:60:33: note: in expansion of macro 'DT_NODELABEL'
   60 |  nrf_gpio_cfg_input(DT_GPIO_PIN(DT_NODELABEL(button0), gpios),
      |                                 ^~~~~~~~~~~~
[92/141] Linking C static library zephyr/lib/posix/liblib__posix.a
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/koutaiba/zephyrproject/zephyr/samples/boards/nrf/system_off/bin`

**Environment (please complete the following information):**
 - OS: ubuntu 
 - Toolchain: Zephyr SDK using VCStudio

Thanks

galak, 2021-07-13:
This is expected as there is no nodelabel in the devicetree for `button0:` that is needed by this sample.

mbolivar-nordic, 2021-07-14:
> This is expected as there is no nodelabel in the devicetree for `button0:` that is needed by this sample.

I also looked at https://github.com/particle-iot/xenon and it doesn't look like there is a button we can use for this sample.

The samples/boards/nrf/system_off/README.rst file explicitly documents nRF51 and nRF52 DKs as supported boards.

So this sample just does not support this board; please use a DK or some other board with a button0 in its devicetree as described above.

closed 2021-07-14 by mbolivar-nordic

koutaiba95, 2021-07-15:
Thank you for your feedback. Apparently i have more work to do to understand differences between boards. The Xenon is listed as equipped with Nordic nRF52840 so i assumed it's compatible.
Any way i hope this thread will help someone else "au cas où".
There is a button on the Xenon and the data tree provided in zephyr at : [https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/particle_xenon/dts/mesh_feather.dtsi](url) has the following code : 

`

		compatible = "gpio-keys";
		mode_button: button_0 {
			gpios = <&gpio0 11 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
			label = "Mode Button";
		};

`
however replacing button0 with button_0 or sw0 didn't work, but exchanging it with mode_button did the trick....

mbolivar-nordic, 2021-07-15:
> The Xenon is listed as equipped with Nordic nRF52840 so i assumed it's compatible.

Right, but nRF52840 is an SoC, not a board. The sample has additional requirements beyond just a Nordic SoC.

> There is a button on the Xenon and the data tree provided in zephyr

Is that a generic button available for application specific use?

I saw there was a mode button on the hardware description page, but the name "mode button" makes it seem like it is not generic, but serves some particular purpose.

If it's reserved for something specific, then I don't think we should treat it like it's generic.

If it is indeed generic,

```
	button0: mode_button: button_0 {
```

can be added to the devicetree to support the sample.

koutaiba95, 2021-07-16:
Hello, 
You are right a SoC is not a board. I haven't thought about the fact that a sample needs a specific board or a bunch of similar boards to be compatible with as the pinout will vary from board to board.

The button is not generic as you mentioned, it serves a purpose in the official software provided with the board as you pointed out, that is where the name comes from.