zephyrproject-rtos/zephyr #36927: LWM2M: Writing to Write-Only resource causes notification

huelsenfruchtzwerg, 2021-07-13:
**Describe the bug**
Writing to any ressource that's declared as writable only triggers a notification, i.e. if the containing object is observed.

**Expected behavior**
Writing to non-readable ressources should not trigger a notification. At least that's how Leshan implements it (... I'm told)

**Impact**
Unnecessary Trafic/Power-Consumption

**Environment (please complete the following information):**
Nordic Connect SDK 2.6.0, nrf9160

rlubos, 2021-07-15:
I agree, it doesn't make much sense to send a notification if there are no new data to notify. I'll send a patch.

rlubos, 2021-07-15:
@huelsenfruchtzwerg Please have a look at https://github.com/zephyrproject-rtos/zephyr/pull/36984, it was actually a pretty simple fix. Feel free to approve if it works ok for you.

PR #36984 (2021-07-20): net: lwm2m: Prevent notifications on non-readable resources
In case a non-readable resource gets updated (either by the server or
with an API), it makes no sense to send a notification in such case, as
no such resources are not included in notifications anyway.

Fixes #36927
closed 2021-07-20 by carlescufi